microsoft/AirSim #433: Upgrading to UE4.17: Possible change needed in camera recording code

saihv, 2017-08-23:
Looks like starting in 4.17, `FImageUtils::CompressImageArray` needs to be called from the game thread: I couldn't find a reference to this in the changelog, so it's probably something underlying that's changed. In 4.17, RenderRequest.cpp throws a `IsInGameThread()=false` exception. I guess [this part](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/RenderRequest.cpp#L67) needs to be wrapped inside a `CreateandDispatchWhenReady` block so that it will be queued up to the game thread.

` FGraphEventRef Task = FFunctionGraphTask::CreateAndDispatchWhenReady([this, &image_data]() {
	FImageUtils::CompressImageArray(data->width, data->height, data->bmp, image_data);
	}, TStatId(), NULL, ENamedThreads::GameThread);

// Wait for the code above to complete do this:
  FTaskGraphInterface::Get().WaitUntilTaskCompletes(Task);
`

If the plan is to make 4.17 the official version from now on, perhaps this addition will be needed..

sytelus, 2017-08-23:
Thanks for pointing this out. I noticed that 4.16 is showing warning instead (on Windows):

`[2017.08.23-11.12.44:597][822]LogModuleManager:Warning: ModuleManager: Attempting to load 'ImageWrapper' outside the main thread.  This module was already loaded - so we didn't crash but this isn't safe.  Please call LoadModule on the main/game thread only.  You can use GetModule or GetModuleChecked instead, those are safe to call outside the game thread.`

So made [this one line change](https://github.com/Microsoft/AirSim/blob/master/Unreal/Plugins/AirSim/Source/AirSimGameMode.cpp#L16).  This still doesn't get rid of above warning. I'll test out above code soon.

abc19850508, 2017-10-15:
I cannot get 4.16 now. can u tell me how could I get the 4.16 version from Epic?

msb336, 2019-10-02:
Closing this issue, as we are now supporting builds 4.18-4.22

closed 2019-10-02 by msb336