microsoft/AirSim #435: How to replace the drone to be a camera

jingleFun, 2017-08-23:
I want to get server images from one position with different orientation, but it seems that the center of rotation is the drone not the camera, am I right? how can I fix it? Actually, I have some positions, I just want to get same images(the depth image and label image) with server fixed orientations,  but The drone sometimes block the camera`s view, So how can I replace the drone with just a camera? Thank you.

sytelus, 2017-08-29:
Good points. You are correct that center of rotation is drone, not camera. Let me think about how we can change this...

jingleFun, 2017-08-29:
I`ve figure this out ^-^. I changed the BP_Flaying Pawn in unreal ,set the drone invisible, and set the camera to (0, 0, 0).

sytelus, 2018-10-17:
The new computer vision mode now exactly does this.

closed 2018-10-17 by sytelus