microsoft/AirSim #436: Clarify best known working configuration

xoltar, 2017-08-24:
As a new user, it is very difficult to determine how to get a working system. I need a realistic physics system. I bought a pixhawk mini solely for this system because the docs said it would work out of the box, which it didn't. 

Please clarify, am I supposed to:

1. Use my pixhawk mini with old PX4 firmware (1.5.4 I guess)
2. Use my pixhawk mini with ROS 
3. Buy some other hardware to use with ROS
4. Something else?

Also, what environment are the developers using? I have found that QGroundControl is completely unable to update firmware on linux, but works fine on Windows, which makes me wonder if AirSim developers are not hitting these problems due to mostly working on Windows?

There are enough moving parts in this system that a documented "best known configuration" would help a lot.

xoltar, 2017-08-24:
I finally got things working. Using Windows definitely helped. The Unreal Editor is also more stable on Windows. I used firmware 1.5.5 with the pixhawk mini. It would help to explain how to get the "HIL Quadcopter" airframe into the QGroundControl list, since it is not always in the list of available airframes, and that appears to be a crucial piece of the setup.

sytelus, 2017-08-29:
You are correct that firmware flushing and UE4 development is much stable on Windows. So we actually use Windows as our main development environment but we make sure that things compiles and runs on Linux as well.

I would not recommend using PX4 unless you are familiar with it. As lot of people have also echoed your sentiment we recently embarked on a mission to create simulator-friendly alternative to PX4. We call it simple_flight and it is now default in latest version. That means you can use AirSim without PX4 and related additional steps! Please be sure to see [our docs](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md) for more details.

closed 2017-08-29 by sytelus