microsoft/AirSim #437: Accelerator jumps periodically in z-axis

mocibb, 2017-08-25:
Hi, i am trying use the latest Airsim and PIX4 firmware(v1.6.5) for HIL simulation.
When uav is on ground, in qgroundcontrol i found accelerator zacc  value jump between 0 and -10.
that cause estimator of PIX4 getting crazy.
I am unable to figure out why this happened, have any ideas about this

[image]

sytelus, 2017-08-29:
I think this was solved using ground lock before. I'll have to re-test and get back to you.

mocibb, 2017-08-30:
I used latest commit(a20728d) for testing.
it seems z-acc still have some problems, but different from previous one.
[image]

I also found altitude have been periodically reset.
[image]

lovettchris, 2017-12-09:
This might be fixed now with [my new pull request](https://github.com/Microsoft/AirSim/pull/670).

lovettchris, 2017-12-12:
Please reopen if you still see this, thanks.

closed 2017-12-12 by lovettchris