microsoft/AirSim #443: How to control the drone accurately?

lxgen, 2017-08-30:
When I call the api `moveByVelocityZ` at fixed frequency, I find that the drone takes a while to reach the desired velocity. And the time it takes is related to its current velocity and my desired velocity. So the next time when I call the api, the drone's current velocity isn't really what I want at the last time. Then I can't control the drone accurately. How can I solve it? Thans you.

sytelus, 2017-09-01:
Are using PX4? Can you give it a try with latest version which uses simple_flight by default (make sure to delete settings.json to use defaults).

xiaolin360, 2017-09-03:
Hello. I tried the latest version of airsim, and i deleted the settings.json, then i startup my UE project, an error occured which showed that "RC Controller on USB not detected:4294967295", why was this, please?

sytelus, 2017-09-17:
That just means you don't have remote control connected to your PC so you can't fly using remote control.

Please see: Please see this doc: https://github.com/Microsoft/AirSim/blob/master/docs/remote_control.md

closed 2017-09-17 by sytelus