microsoft/AirSim #445: Unexpected 'f' Expected '}' Error In Settings When Running AirSim

exploke, 2017-08-31:
When I run the latest commit of AirSim, I get an unexpected 'f' expected '}' error in the settings.json file.

[image]

The error in the settings file seems to be with RecordInterval portion.

[image]

Any help to fix this problem as soon as possible would be appreciated.

Robin-Wong, 2017-09-01:
Please try to delete the 'f' after all the float value,
like change ""AutoExposureMaxBrightness": 0.64f" to ""AutoExposureMaxBrightness": 0.64".
By the way, add ',' in line 20("ImageType":1,), 35, 43.

sytelus, 2017-09-01:
Also, if setting is just default value, I would highly recommand not adding it in setting.json. This way if default changes in later releases, you get it without issues.

closed 2017-09-01 by sytelus

Curry30h, 2019-03-21:
I also encountered the same problem. Is there a solution?