microsoft/AirSim #446: CameraActor and BP_FlyingPawn position not align in sim mode

jingleFun, 2017-09-01:
I am running a sample code with CubicImageGenerator.hpp , but when I set the pose to (0, 0, 0) , The CameraActor Transform Location turn to (0, 0, 0)， but BP_FlyingPawn turn to (5690, -100, 202). How is that? I want to set the drone position to (0, 0, 0) ,but it seems that the view is not from (0, 0, 0), more like to be (5690, -100, 202). How can fix it? Thank you !

sytelus, 2017-09-01:
Locations you get are in NED frame. In NED frame your start location is 0,0,0. In environment start location would be actual coordinates. Also notice that Z is reversed in NED frame.

closed 2017-09-01 by sytelus

jingleFun, 2017-09-07:
But how can I use simSetPos funtion to trans a position like(0, 0, 0) then I can  move the drone to the right place (0, 0, 0) ? And what is actual coordinates?  Do you mean longitude , latitude or something else?