microsoft/AirSim #447: how to build Binary environment

fmzhu, 2017-09-01:
hello,
so how to build binary environment.  I've tried the"cook" function or "package" function from the unreal engine, but they didn't work. Thank you very much for your concern.

saihv, 2017-09-02:
Those should be the only two steps required, on Windows at least. I used the same two functions to create binaries with custom environments etc. Are you getting any specific error messages while trying to cook the content?

sytelus, 2017-09-03:
Agreed with @saihv. Just go to File > Package in UE Editor. I usually select DebugGame but other config seems to work as well. Please include detailed error messages you get.

sytelus, 2017-09-05:
Also see Unreal Engine docs:
https://docs.unrealengine.com/latest/INT/Engine/Basics/Projects/Packaging/
https://docs.unrealengine.com/latest/INT/Engine/Deployment/

Best way to get support for this is ask questions in Unreal forums: http://answers.unrealengine.com

closed 2017-09-05 by sytelus