microsoft/AirSim #449: Cannot Change Camera Resolution

abe-mart, 2017-09-01:
The documentation gives the following example settings file to change camera settings.

```
{
  "SceneCaptureSettings" : {
    "Width": 256,
    "Height": 144,
    "FOV_Degrees": 90,
    "AutoExposureSpeed": 100,
    "MotionBlurAmount": 0
  },
  "DefaultVehicleConfig": "SimpleFlight",
  "UsageScenario": "ComputerVision"
}
```
However, changing the 'Width' and 'Height' values seems to have no effect on the images produced.  This is true both for the built in recording function, and for images taken through the Python API.

I'd like to take larger images than the default settings.

saihv, 2017-09-02:
I've had that happen to me before if there is a syntax error elsewhere in the JSON file. Could you make sure there are no such things like trailing commas, and check whether AirSim complains about the settings.json when the simulation first starts? (Error messages are shown in the top left)

sytelus, 2017-09-03:
We had to change the schema for capture settings to accomodate many new views we have added. Specifically, instead of SceneCaptureSettings, DepthCaptureSettings etc we now just have CaptureSettings with new element "ImageType". So your above settings would be as follows in new scheme of things:

```
"CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "MotionBlurAmount": 0    }
  ]
```

The available ImageType are as follows:
```
                Scene = 0, 
                DepthMeters = 1, 
                DepthVis = 2, 
                DisparityNormalized = 3,
                Segmentation = 4,
                SurfaceNormals = 5,
```

Sorry about the breaking change but we think this will give more flexibility as we add more image types.

abe-mart, 2017-09-05:
Perfect, that fixes the problem, thanks!  I copied my broken settings from the [Image APIs page](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#changing-resolution-and-camera-parameters), so that section will probably need to be updated as well to reflect the new settings.

sytelus, 2017-09-05:
Thanks for pointing this out. I've fixed doc as well now.

closed 2017-09-05 by sytelus

chaitanyantr, 2018-11-27:
the output images from after recording (Press R in unreal project)....and by API Python call the image capture resolution is 256x144 even i changed the json file to 2000x1000 

"CaptureSettings": [
    {
      "ImageType": 0,
      "Width": 2000,
      "Height": 1000,
      "FOV_Degrees": 90,
      "MotionBlurAmount": 0    
    }
  ]