microsoft/AirSim #451: Weird little bug in camera director: manual camera control

saihv, 2017-09-02:
When using the manual camera setting feature (by pressing M): there is a strange bug occurring that affects the subwindow layout in the HUD. As soon as M is pressed to enter this mode, the leftmost container is flipped and pushed to the left. This happens even if that container is moved anywhere within the HUD.

[image]

This seems to be a side effect when the code executes the `enableBindings()` function within ManualPoseController.cpp. If I, say, comment out all those lines in the function, this bug disappears, and manual camera control still works fine.

sytelus, 2017-09-03:
I can reproduce this as well. I'm still not sure what is the connection with enableBindings(). I guess we need to figure out workaround...

rajat2004, 2021-01-19:
This doesn't seem to happen any longer, can be closed I think?

saihv, 2021-01-24:
Fixed in later releases, closing.

closed 2021-01-24 by saihv