microsoft/AirSim #455: UnrealEngine/Engine/Source/ThirdParty/Linux/LibCxx/include/c++/v1/cstddef:43:15: fatal error: 'stddef.h'  file not found  #include_next <stddef.h>

fenfeibani, 2017-09-04:
**# When i compile the UnrealEngine4.17 source code , both the ./Setup.sh and ./GenerateProjectFiles.sh are successful, but when type "make" command, the message is as follow:**

ddname-pc:~/UnrealEngine$ make
bash "/home/ddname/UnrealEngine/Engine/Build/BatchFiles/Linux/Build.sh" CrashReportClient Linux Shipping  
Building CrashReportClient...
Creating makefile for CrashReportClient (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
Building UnrealHeaderTool...
Creating makefile for UnrealHeaderTool (no existing makefile)
Performing full C++ include scan (no include cache file)
Using clang version '3.8.0' (string), 3 (major), 8 (minor), 0 (patch)
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Performing 25 actions (12 in parallel)
[3/25] Compile SharedPCH.Core.h
[4/25] Compile Module.UnrealHeaderTool.cpp
[1/25] Compile PCH.Core.h
[2/25] Compile PCH.CoreUObject.h
In file included from /home/ddname/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UnrealHeaderTool/Development/Core/PCH.Core.h:46:
In file included from /home/ddname/UnrealEngine/Engine/Source/Runtime/Core/Private/CorePrivatePCH.h:10:
In file included from Runtime/Core/Public/GenericPlatform/GenericPlatformMisc.h:7:
In file included from Runtime/Core/Public/HAL/PlatformCrt.h:5:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/new:70:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/exception:81:
ThirdParty/Linux/LibCxx/include/c++/v1/cstddef:43:15: fatal error: 'stddef.h'
      file not found
#include_next <stddef.h>
              ^
In file included from /home/ddname/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UnrealHeaderTool/Development/Core/SharedPCH.Core.h:39:
In file included from /home/ddname/UnrealEngine/Engine/Source/Runtime/Core/Public/CoreSharedPCH.h:7:
In file included from Runtime/Core/Public/Misc/Exec.h:6:
In file included from Runtime/Core/Public/Misc/AssertionMacros.h:6:
In file included from Runtime/Core/Public/HAL/PlatformMisc.h:5:
In file included from Runtime/Core/Public/GenericPlatform/GenericPlatformMisc.h:7:
In file included from Runtime/Core/Public/HAL/PlatformCrt.h:5:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/new:70:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/exception:81:
ThirdParty/Linux/LibCxx/include/c++/v1/cstddef:43:15: fatal error: 'stddef.h'
      file not found
#include_next <stddef.h>
              ^
In file included from /home/ddname/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UnrealHeaderTool/Development/CoreUObject/PCH.CoreUObject.h:42:
In file included from /home/ddname/UnrealEngine/Engine/Source/Runtime/CoreUObject/Private/CoreUObjectPrivatePCH.h:6:
In file included from Runtime/Core/Public/CoreMinimal.h:26:
In file included from Runtime/Core/Public/HAL/PlatformCrt.h:5:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/new:70:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/exception:81:
ThirdParty/Linux/LibCxx/include/c++/v1/cstddef:43:15: fatal error: 'stddef.h'
      file not found
#include_next <stddef.h>
              ^
In file included from /home/ddname/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/UnrealHeaderTool/Development/UnrealHeaderTool/Module.UnrealHeaderTool.cpp:2:
In file included from /home/ddname/UnrealEngine/Engine/Source/Programs/UnrealHeaderTool/Private/BaseParser.cpp:3:
In file included from Programs/UnrealHeaderTool/Private/BaseParser.h:5:
In file included from Runtime/Core/Public/CoreMinimal.h:26:
In file included from Runtime/Core/Public/HAL/PlatformCrt.h:5:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/new:70:
In file included from ThirdParty/Linux/LibCxx/include/c++/v1/exception:81:
ThirdParty/Linux/LibCxx/include/c++/v1/cstddef:43:15: fatal error: 'stddef.h'
      file not found
#include_next <stddef.h>
              ^
1 error generated.
1 error generated.
1 error generated.
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /home/ddname/UnrealEngine/Engine/Binaries/Linux/libUnrealHeaderTool-Core.so
Total build time: 3.71 seconds (Local executor: 0.00 seconds)
Error: UnrealHeaderTool failed for target 'CrashReportClient' (platform: Linux, module info: /home/ddname/UnrealEngine/Engine/Intermediate/Build/Linux/B4D820EA/CrashReportClient/Shipping/CrashReportClient.uhtmanifest, exit code: OtherCompilationError (5)).
Makefile:252: recipe for target 'CrashReportClient-Linux-Shipping' failed
make: *** [CrashReportClient-Linux-Shipping] Error 5

sytelus, 2017-09-04:
Are you using Ubuntu 16.04 LTS? I'd seen one build error when someone in Unreal team made a bad checkin in 4.17 branch. They fixed it in 3-4 hours. I would suggest to post this in https://answers.unrealengine.com. and/or get previous commits.

fenfeibani, 2017-09-04:
@sytelus
yes,my system is Ubuntu 16.04 LTS. Thank you for your suggestion!

closed 2017-09-05 by sytelus