microsoft/AirSim #456: Camera Gimbal / Separate Camera and Drone Poses

abe-mart, 2017-09-05:
I'd like to try using AirSim to simulate UAV mapping/photogrammetry flights.  Is there a way to control the camera pose separately from the vehicle pose?

sytelus, 2017-09-05:
Currently no but soon we are planning to revamp APIs and this would be a good feature to add. Its actually fairly easy to add as well so please feel free to contribute :).

sytelus, 2018-02-21:
I just added Gimbal APIs in AirSim. [Here's doc](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#camera-orientation-gimble).

abe-mart, 2018-03-07:
This is excellent.  However, when I try to use the new command in Python with the latest version of Airsim and the Landscape Mountains binary, I get the following error:

`RPCError: rpclib: server could not find function 'setCameraOrientation' with argument count 2.`

This was with the command copied and pasted from the docs:

`client.setCameraOrientation(0, AirSimClientBase.toQuaternion(0.261799, 0, 0)); #radians`

guillem74, 2018-03-15:
I get the same error:

RPCError: rpclib: server could not find function 'setCameraOrientation' with argument count 2.

sytelus, 2018-04-25:
You will need to use latest binaries from release 1.1.10.

abe-mart, 2018-04-30:
Thanks, works great with the latest binaries.

closed 2018-04-30 by abe-mart