microsoft/AirSim #458: Frame rate of camera.py

ZixingLin, 2017-09-07:
I'm trying to get a consistent video stream from AirSim via camera.py, however the frame rate of the video is almost always below 10 FPS, even worse when trying to connect with a remote device on the network. Is there a way to increase the frame count/rate of image received or is it hardware based (I'm not running a very powerful machine)?

sytelus, 2017-09-09:
What image resolution are you using? Default is I think 255 X 143 or so. For that size things should be pretty fast (I can easily get 30 FPS on my m/c). If you are using large environment, it might drop a bit. On my laptop I can get 10 FPS however.

ZixingLin, 2017-09-15:
Ah explains it - I doubled the original resolution hence the lower frame rate. Thank you for your response!

closed 2017-09-17 by sytelus