microsoft/AirSim #460: The project could not be compiled. would you like to open it in an IDE to edit source code?

fenfeibani, 2017-09-08:
When i use UE4Editor open the "AirSim/Unreal/Environments/Blocks/Blocks.uproject",  after generating source code , on the "compiling c++ source code" step, it prompted that "The project could not be compiled. would you like to open it in an IDE to edit source code? ".
Using  bundled libc++ standard c++ library.
Using fast way to relink circularly dependent libraries(no FixDeps).
Performing 6 actions(12 in parallel)
[2/6]Compile Module.AirSim.gen.cpp
[1/6]Compile Module.AirSim.cpp
[3/6]Compile PCH.Blocks.h
[4/6]Compile Blocks.cpp
ERROR:UBT ERROR:Failed to produce item:.../AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

My system is ubuntu16.04,  UE4.17

shristi945, 2017-09-09:
same problem got stuck with this.

sytelus, 2017-09-09:
Just tried out getting latest and building and it seems to work without any compile error on my machine. I'm using following commit for Unreal engine code:

```
~/.../UnrealEngine$ git log -n 5
commit af96417313a908b20621a443175ba91683c238c8
Author: Dan Oconnor <Dan.OConnor@epicgames.com>
Date:   Tue Aug 29 14:46:30 2017 -0400

```

So I would first suggest to try out above commit,

`git checkout af96417313a908b20621a443175ba91683c238c8`

If you still get compile error, could you please attach full log? This log should be available in Saved folder under project as well.

fenfeibani, 2017-09-16:
Follow your steps,i have compiled the c++ source code successfully! Thank you very much!

closed 2017-09-17 by sytelus