microsoft/AirSim #461: the 3 default small windows for depth etc .... do not want to show up

wifibot78, 2017-09-10:
Hello,

I use unreal engine 4.16.3 with a git clone of airsim from yesterday, 
I am able to compile everything and launch the project, not PIP display ?
It does the same with blocs example or landscape example.
I am planing to test HIL with px4 as all my drones uses this flight controller.

Best Regards,

Laurent

sytelus, 2017-09-10:
Press 0 key to show the subwindows. By default they are not shown to avoid perf impact on low end machines.

wifibot78, 2017-09-11:
I had already pressed the 0 key (saw this on the F1 help menu), nothing was coming ?
I use the default settings.jeson (no options), I had also tryed to add :

"SubWindows": [
    {"Index": 1, "ImageType": 3},
    {"Index": 2, "ImageType": 5}

Laurent

wifibot78, 2017-09-16:
ok I have found that in simhud.cpp , EKeys::Zero is not handled on my computer ? If I replace Zero by a letter or a F key it is ok, the 3 windows are displayed.

sytelus, 2017-09-17:
That strange. Are you using non-english keyboard? Or any special keyboard?

wifibot78, 2017-09-18:
yes a laptop azerty keyboard with a numpad. I can test with an English keyboard.

closed 2017-09-19 by sytelus