microsoft/AirSim #462: VS 15 Build Errors

eanmikale, 2017-09-11:
Since the most recent updates to 16 have been having issues with new build in VS 15. Any assistance would be appreciated problem-solving this build. Not sure where the missing files might be. Please see below. 

[image]
[image]

sytelus, 2017-09-12:
Could you please copy-paste text from Output window?

eanmikale, 2017-09-12:
Here is a copy of the output log.... Thanks! 

1>------ Build started: Project: LandscapeMountains, Configuration: DebugGame_Editor x64 ------
1>  Performing full C++ include scan (hot reloading a new target)
1>  Compiling game modules for hot reload
1>  Parsing headers for LandscapeMountainsEditor
1>    Running UnrealHeaderTool "C:\Users\info\Documents\Unreal Projects\LandscapeMountains\LandscapeMountains.uproject" "C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Intermediate\Build\Win64\LandscapeMountainsEditor\DebugGame\LandscapeMountainsEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>  Reflection code generated for LandscapeMountainsEditor in 4.557897 seconds
1>  Performing 11 actions (9 in parallel)
1>  [2/11] Resource PCLaunch.rc
1>  [3/11] Resource PCLaunch.rc
1>  SharedPCH.Engine.NonOptimized.cpp
1>  LandscapeMountains.cpp
1>  Module.AirSim.cpp
1>  MyClass.cpp
1>  AirSim.generated.cpp
1>  [8/11] Link UE4Editor-LandscapeMountains-1000-Win64-DebugGame.lib
1>     Creating library C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-LandscapeMountains-1000-Win64-DebugGame.lib and object C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-LandscapeMountains-1000-Win64-DebugGame.exp
1>C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\PIPCamera.h(6): fatal error C1083: Cannot open include file: 'controllers/VehicleCameraBase.hpp': No such file or directory
1>C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Source\AirBlueprintLib.cpp(11): fatal error C1083: Cannot open include file: 'common/common_utils/Utils.hpp': No such file or directory
1>ERROR : UBT error : Failed to produce item: C:\Users\info\Documents\Unreal Projects\LandscapeMountains\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-185-Win64-DebugGame.lib
1>  Total build time: 27.28 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" LandscapeMountainsEditor Win64 DebugGame "C:\Users\info\Documents\Unreal Projects\LandscapeMountains\LandscapeMountains.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

eanmikale, 2017-09-13:
Here is a more recent build with the same errors essentially. It seems as if the plugins are not completely accessible during the build. See below...

1>------ Build started: Project: MyProject, Configuration: DebugGame_Editor x64 ------
1>  Performing full C++ include scan (hot reloading a new target)
1>  Compiling game modules for hot reload
1>  Parsing headers for MyProjectEditor
1>    Running UnrealHeaderTool "C:\Users\Ean\Documents\Unreal Projects\MyProject\MyProject.uproject" "C:\Users\Ean\Documents\Unreal Projects\MyProject\Intermediate\Build\Win64\MyProjectEditor\DebugGame\MyProjectEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>  Reflection code generated for MyProjectEditor in 5.7142026 seconds
1>  Performing 12 actions (2 in parallel)
1>  [2/12] Resource PCLaunch.rc
1>  SharedPCH.Engine.NonOptimized.cpp
1>  [3/12] Resource PCLaunch.rc
1>  Module.AirSim.cpp
1>  AirSim.generated.cpp
1>C:\Users\Ean\Documents\Unreal Projects\MyProject\Plugins\AirSim\Source\PIPCamera.h(6): fatal error C1083: Cannot open include file: 'controllers/VehicleCameraBase.hpp': No such file or directory
1>C:\Users\Ean\Documents\Unreal Projects\MyProject\Plugins\AirSim\Source\AirBlueprintLib.cpp(11): fatal error C1083: Cannot open include file: 'common/common_utils/Utils.hpp': No such file or directory
1>  MyProject.cpp
1>  MyProjectGameModeBase.cpp
1>  MyProject.generated.cpp
1>  [9/12] Link UE4Editor-MyProject-7679-Win64-DebugGame.lib
1>     Creating library C:\Users\Ean\Documents\Unreal Projects\MyProject\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-MyProject-7679-Win64-DebugGame.lib and object C:\Users\Ean\Documents\Unreal Projects\MyProject\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-MyProject-7679-Win64-DebugGame.exp
1>ERROR : UBT error : Failed to produce item: C:\Users\Ean\Documents\Unreal Projects\MyProject\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-9029-Win64-DebugGame.lib
1>  Total build time: 22.70 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" MyProjectEditor Win64 DebugGame "C:\Users\Ean\Documents\Unreal Projects\MyProject\MyProject.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========

sytelus, 2017-09-17:
I have never seen this errors. Are you following build instructions properly? See here: https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md

My guess is that you don't have copy of AirSim folder under Plugins folder properly. As you are using your own environment, please make sure to follow these steps: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_custenv.md

The errors are these:
```
1>C:\Users\Ean\Documents\Unreal Projects\MyProject\Plugins\AirSim\Source\PIPCamera.h(6): fatal error C1083: Cannot open include file: 'controllers/VehicleCameraBase.hpp': No such file or directory
1>C:\Users\Ean\Documents\Unreal Projects\MyProject\Plugins\AirSim\Source\AirBlueprintLib.cpp(11): fatal error C1083: Cannot open include file: 'common/common_utils/Utils.hpp': No such file or directory
```

So try to see if those files exists.

eanmikale, 2017-09-25:
Sytelus - We have followed the following steps: 

1) Install cmake
2) Install eigen 
3) Install Boost
4) Place AirSim Plugin in "My  Project" folder
5) Update uproject Plugin
6) Run sln using VS15 w/ Debug Game Editor + Win64
7) Made sure we have the AirSim/External/Rcpclib folder
8) Checked the AirSim.uplugin file
9) Added the update_airsim.cmd file to My Project
10) Added Derived Datacache, Contents, and Config files from Landscape Mountains
11) Under system variables, added AIRSIM repo location, Eigen_Root, Boost, and Cmake. Not sure with the newer updates whether the system variables only need the AIRSIM and Eigen root? 

Errors: The two errors above are still perplexing. The PIPCamera.h file and the AirBlueprintLib.cpp file are both located in the Source folder.  We have a class of students working on this issue, and have been able to use Computer Vision as a work around with Binaries, but would like the class to use more "Drone Shell", which I haven't been able to use since Unreal v.  15. Thanks again. 

And when we ;build,

sytelus, 2017-09-28:
You don't need first 3 steps. I'm guessing you are using old build instructions with new code. Please get latest code and follow the [build instructions](https://github.com/Microsoft/AirSim/blob/master/docs/build_windows.md). In short,

1. Run build from VS x64 Command Prompt in AirSim folder.
2. Copy Plugins\AirSim folder to your Unreal project.
3. Modify uproject to load AirSim plugin.

That's all.

eanmikale, 2017-09-28:
What version of VS? 15 or 17? Some of the instructions say 17 while others say 15. Thank you! Ean Mikale, J.D.Chief Innovation OfficerInfinite 8 Institute, L3Cean@infinite8institute.comwww.infinite8institute.comU.S. Federal Contractor---- On Thu, 28 Sep 2017 04:17:59 -0500  Shital Shah<notifications@github.com> wrote ----You don't need first 3 steps. I'm guessing you are using old build instructions with new code. Please get latest code and follow the build instructions. In short,  Run build from VS x64 Command Prompt in AirSim folder. Copy PluginsAirSim folder to your Unreal project. Modify uproject to load AirSim plugin.  That's all.

sytelus, 2017-09-28:
Please use VS 2015. I don't think any docs asks to use VS 2017. Where did you see it?

Mayankm96, 2017-10-01:
Well, it's working now. Thank you for fixing this, @sytelus ! :)

closed 2017-10-04 by sytelus