microsoft/AirSim #463: Binary build: hello_drone.py: No such file or directory: '\\temp\\py1.png'

Kjell-K, 2017-09-14:
Hey Folks,
The Binary build works just fine and I am able to test a lot of the python APIs, but the hello_drone.py is not able to take a images. I get the following error message: 

```
  File "C:/Users/Kjell/Desktop/Test/PythonClient/hello_drone.py", line 27, in <module>
    AirSimClient.write_file(os.path.normpath('/temp/py1.png'), response.image_data_uint8)

  File "C:\Users\Kjell\Desktop\Test\PythonClient\PythonClient.py", line 231, in write_file
    with open(filename, 'wb') as afile:

FileNotFoundError: [Errno 2] No such file or directory: '\\temp\\py1.png'
```
Changing '/temp/py1.png' to '\\temp\\py1.png' in hello_drone does not work. 

Thanks in advance

lovettchris, 2017-09-14:
You appear to be on Windows, so just try  AirSimClient.write_file("py1.png") so it writes to whatever your current directory is.  I'm guessing it can't find a "temp" folder in the root of your "current drive".  You could also try "c:/temp/py1.png" to specify the drive.  Note: Python on Windows can handle forward slashes.

Kjell-K, 2017-09-14:
Yes it could not find the temp folder. 
A simple dot does the trick.
`'./temp/py1.png'`
Thank you.

closed 2017-09-14 by Kjell-K

harnix, 2020-03-16:
Hello, @kjell-k would appreciate it if you could give details on how you solved this issue. Did you manually create the temp directory?