microsoft/AirSim #466: Known good dev environment?

n1ckfg, 2017-09-15:
Having trouble compiling the Unreal plugin everywhere--in Ubuntu 16.04 and Win10, VS 2017 and VS 2015, Unreal 4.16 and 4.17. What's a best-practice dev environment (OS, Unreal version, VS version, etc.) to test against so I can figure out what's going wrong?

saihv, 2017-09-16:
Works great for me on Windows 10, Unreal 4.16.3 and Visual Studio 2015 Community Update 3.

sytelus, 2017-09-17:
I'm able to compile from scratch on Win 10 as well as Ubuntu 16.04 LTS. What is the issue you are facing? Please post complete error messages and/or log files.

n1ckfg, 2017-09-18:
Starting with @saihv's recommended environment, build fails with:
```
1>C:\Users\nick\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\PIPCamera.h(6): fatal error C1083: Cannot open include file: 'controllers/VehicleCameraBase.hpp': No such file or directory
1>C:\Users\nick\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirBlueprintLib.cpp(11): fatal error C1083: Cannot open include file: 'common/common_utils/Utils.hpp': No such file or directory
1>  Blocks.cpp
1>  [9/12] Link UE4Editor-Blocks.lib
1>     Creating library C:\Users\nick\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Blocks.lib and object C:\Users\nick\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-Blocks.exp
1>ERROR : UBT error : Failed to produce item: C:\Users\nick\Documents\GitHub\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\Development\UE4Editor-AirSim.lib
```

n1ckfg, 2017-09-18:
Also fails with a very similar error in Ubuntu 16.04 with both UE 4.16 and UE 4.17:
ERROR: UBT ERROR: Failed to produce item: ~/Projects/Wise/AirSim/Unreal/Environments/Blocks 4.16/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

ERROR: UBT ERROR: Failed to produce item: ~/Projects/Wise/AirSim/Unreal/Environments/Blocks 4.17/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so

I must be making the same mistake in all four scenarios (Win10/VS 2015/UE 4.16, Win10/VS 2017, UE 4.17, Ubuntu 16.04/UE 4.16, Ubuntu 16.04/UE 4.17)

sytelus, 2017-09-19:
Are you following this instructions? https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md

Specifically step #2, which will copy Plugins/AirSim folder in Blocks project. Above errors seems to indicate you don't have this folder.

n1ckfg, 2017-09-19:
Turns out I can't complete the pre-build step for rpclib--CMake complains:  
```
No CMAKE_C_COMPILER could be found.
No CMAKE_CXX_COMPILER could be found.
```
Experimented with various configurations for VS 2015, VS 2017, and standalone CMake install as per docs...any suggestions?

n1ckfg, 2017-09-19:
OK, found the issue: when both VS 2015 and VS 2017 are installed, build.cmd must be run with VS 2017. The Unreal project can still optionally be built with VS 2015. I see the VS 2017 flag for cmake is already provided in build.cmd and commented out; would be great to add an explainer note in the docs. Thanks for the suggestions!

closed 2017-09-19 by n1ckfg

spencerh-b, 2017-09-20:
So the fix was uncommenting?

n1ckfg, 2017-09-20:
Yeah afaik if you have both VS 2017 and VS 2015 installed, you _must_ run build.cmd for VS 2017 (by uncommenting the VS 2017 CMake command string and commenting out the VS 2015 one). After you've built the plugin, you can still build your Unreal project with either VS 2015 or VS 2017.

n1ckfg, 2017-09-20:
I'm still stuck on compiling on Ubuntu for both 4.16 and 4.17.

spencerh-b, 2017-09-22:
@n1ckfg Sweet I will try this

olivia-skydio, 2018-01-24:
@n1ckfg Were you ever able to compile on Ubuntu 16.04? Stuck on this same issue now.

n1ckfg, 2018-01-25:
Yup, for Unreal 4.17. Where does it break for you?

olivia-skydio, 2018-01-25:
When I run Unreal Engine and try and pick a project it says that Airsim isn't compatible 
[image]

I click No to disabling it and then it says: 
[image]

Then if I try to rebuild I get 
[image]

n1ckfg, 2018-01-25:
So afaik you can't build the AirSim plugin in a new project from UE--you have build it in the AirSim repo first and then copy from the AirSim/Unreal folder to your own project.  Next delete Binaries, Intermediate, and Saved folders from your project (this will all be recreated). Then run GenerateProjectFiles.sh in the UE folder on your new.uproject file. Finally try opening the project in Unreal again and see if it builds.

olivia-skydio, 2018-02-02:
How do you build  in the AIrSim repo? Is that just running ./build.sh ? Should that create a .so file?

n1ckfg, 2018-02-03:
First run setup, then build.