microsoft/AirSim #468: How to fly this drone in  "AirSim/Unreal/Environments/Blocks/Blocks.uproject" ?

fenfeibani, 2017-09-16:
After i open the "AirSim/Unreal/Environments/Blocks/Blocks.uproject" in UE4Editor, i can see a quadrotor drone, but i can not fly it. How to fly this drone?

sytelus, 2017-09-17:
Please see this doc: https://github.com/Microsoft/AirSim/blob/master/docs/remote_control.md

To fly manually, you need remote control or RC. If you don't have one then you can use [APIs](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md) to fly programmatically or use so-called [Computer Vision mode](https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md) to move around using keyboard.

closed 2017-09-17 by sytelus

Dola47, 2018-11-28:
@sytelus  Hello,
Can you illustrate to me why Drone does not fly. As far as I see, AirSim contains a HelloDrone file that should allow the drone to fly using the keyboard and it is stated on AirSim Unity page that we should press the arrows or WASD + Page up and Page down to control the drone from the keyboard.

I wonder, why it is does not need to move? Moreover, the cameras works correctly so when I press 0,1,2,3 camera views changes. However, also the car moves correctly.

Please make it a little a bit clear to me.

Thanks

jasonof75, 2020-02-17:
Hello,
@Dola47 have you been able to control the drone via APIs as instructed? 
I am asking because I am running onto the same problem and I cannot find direct instructions regarding the process of installing and using these APIs.

Thanks!

AIdev014, 2020-04-30:
hello,
@jasonof75 i can call Api's using python script. 

import airsim

# connect to the AirSim simulator
client = airsim.MultirotorClient()
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)

# Async methods returns Future. Call join() to wait for task to complete.
client.takeoffAsync().join()

XinXu1999, 2020-06-17:
sorry, I am a beginner in coding and in my case, using the keyboard command does not allow the drone to fly. I am confused to how to set up the RC control of use python APIs. Can anyone explain more in detail?

nati1948, 2020-06-18:
hi,
Try to follow the next steps
1. build airsim (i ran the build.cmd command from the VS devloper command prompt)
2. run the BLOCKS environment and see that unreal environment is shown
3. get into the unreal and press  the **play**  button (or cliclk alt +p )
4. choose the "no" for the drone option 
5. get back to your python code and run "hello_drone"
6. follow the messages on the runing console
if it still not flying , try to remark all the lines using the method : airsim.wait_key("press any key to takeoff")

gary16, 2024-02-26:
> @sytelus Hello, Can you illustrate to me why Drone does not fly. As far as I see, AirSim contains a HelloDrone file that should allow the drone to fly using the keyboard and it is stated on AirSim Unity page that we should press the arrows or WASD + Page up and Page down to control the drone from the keyboard.
> 
> I wonder, why it is does not need to move? Moreover, the cameras works correctly so when I press 0,1,2,3 camera views changes. However, also the car moves correctly.
> 
> Please make it a little a bit clear to me.
> 
> @fenfeibani Have you been able to solve this issue?
> 
> Thanks

if my laptop doesnt have pageup and pagedown . and what will happend?