microsoft/AirSim #469: 'ERROR: UBT ERROR: Failed to produce item' when loading project into Unreal Editor

pratyushmore, 2017-09-17:
Hi, I am working on a project involving training drones in a simulated environment. I am planning to use AirSim for this project. However, I am unable to load the Blocks environment into Unreal Editor. I would severely appreciate any help!

I am using a 2017 Macbook Pro with MacOS Sierra, and Unreal Engine 4.17. When I browse for and load the Blocks project, I get the following error on XCode (after choosing to Convert In-Place):

```
[2017.09.17-07.32.58:117][140]/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:83:13: error: use of undeclared identifier 'v'
[2017.09.17-07.32.58:117][140]    convert(v);
[2017.09.17-07.32.58:117][140]            ^
[2017.09.17-07.33.00:795][140]In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:23:
[2017.09.17-07.33.00:795][140]/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:125:10: fatal error: 'libudev.h' file not found
[2017.09.17-07.33.00:795][140]#include <libudev.h>
[2017.09.17-07.33.00:795][140]         ^
[2017.09.17-07.33.01:312][140]18 errors generated.
[2017.09.17-07.33.05:339][140][4/362] Compile Blocks.cpp
[2017.09.17-07.33.10:006][140]ERROR: UBT ERROR: Failed to produce item: /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib
[2017.09.17-07.33.10:211][140]Total build time: 149.46 seconds (Local executor: 0.00 seconds)
```

This is the error from the Unreal Editor Error Console:

```
The project could not be compiled. Would you like to open it in Xcode?

Running Mono...

Setting up Mono
~/code/capstone/UnrealEngine/Engine ~/code/capstone/UnrealEngine/Engine/Binaries/Mac
@progress push 5%
Building UnrealHeaderTool...
Target is up to date
Deploying UnrealHeaderTool Mac Development...
Deploying now!
Total build time: 3.11 seconds (NoActionsToExecute executor: 0.00 seconds)
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Mac/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
Reflection code generated for BlocksEditor in 12.8137727 seconds
@progress pop
Performing 362 actions (4 in parallel)
[3/362] Compile Module.AirSim.gen.cpp
[2/362] Compile PCH.Blocks.h
[1/362] Compile Module.AirSim.cpp
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:3:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/src/api/RpcLibClient.cpp:19:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/client.h:9:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack.hpp:23:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/object.hpp:835:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/type.hpp:12:
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:32:1: error: declaration of anonymous struct must be a definition
struct nil { };
^
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:34:23: error: expected parameter declarator
inline bool operator<(nil const& lhs, nil const& rhs) {
                      ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/MacTypes.h:92:19: note: expanded from macro 'nil'
      #define nil nullptr
                  ^
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:3:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/src/api/RpcLibClient.cpp:19:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/client.h:9:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack.hpp:23:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/object.hpp:835:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/type.hpp:12:
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:34:23: error: expected ')'
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/MacTypes.h:92:19: note: expanded from macro 'nil'
      #define nil nullptr
                  ^
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:34:22: note: to match this '('
inline bool operator<(nil const& lhs, nil const& rhs) {
                     ^
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:34:13: error: overloaded 'operator<' must have at least one parameter of class or enumeration type
[... 83 lines cut ...]
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/type.hpp:12:
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:82:27: error: expected unqualified-id
    clmdep_msgpack::type::nil v;
                          ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX10.12.sdk/usr/include/MacTypes.h:92:19: note: expanded from macro 'nil'
      #define nil nullptr
                  ^
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:3:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/src/api/RpcLibClient.cpp:19:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/client.h:9:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack.hpp:23:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/object.hpp:835:
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/type.hpp:12:
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/AirLib/deps/rpclib/include/rpc/msgpack/adaptor/nil.hpp:83:13: error: use of undeclared identifier 'v'
    convert(v);
            ^
In file included from /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:23:
/Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:125:10: fatal error: 'libudev.h' file not found
#include <libudev.h>
         ^
18 errors generated.
[4/362] Compile Blocks.cpp
ERROR: UBT ERROR: Failed to produce item: /Users/pratyushmore/code/capstone/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib
Total build time: 149.46 seconds (Local executor: 0.00 seconds)
```

sytelus, 2017-09-19:
The compile error seems to be in rpclib. I've files issue at their github: I’ve filed issue at rpclib: https://github.com/rpclib/rpclib/issues/154

sztomi, 2017-09-21:
I commented on rpclib/rpclib#154 @pratyushmore

pratyushmore, 2017-09-27:
@styelus @sztomi Sorry for the late response - I was out of town. I will take a look at this now and let you know whether it worked out! Thanks!

pratyushmore, 2017-09-27:
Hi. I built the specified commit from the rpclib repo. (Do I have to rebuild AirSim or Unreal Engine or link this new rpclib installation somehow?)

Now, when I try to import the Blocks project into unreal editor, I get the following message:

```
This project requires the 'AirSim' plugin, which is not compatible with the current engine version.

Would you like to disable it? You will no longer be able to open any assets created using it.
```
I choose No. I get the following

```
The following modules are missing or built with a different engine version:

UE4Editor-Blocks.dylib
UE4Editor-AirSim.dylib

Would you like to rebuild them now?
```

I choose Yes. 

The Editor Crashes and a dialogue box says I should try rebuilding from source.

pratyushmore, 2017-09-30:
@sztomi Any tips on how I can proceed from here?

BoyuanChen, 2017-09-30:
@pratyushmore  I just fixed it by reinstall libudev. If you do not have it before, try this one:
```
sudo apt-get install libudev-dev
```
Hope this can help.

pratyushmore, 2017-10-01:
@BoyuanChen Hi. I have not yet fixed this. I will try to find the equivalent of libudev for macOS and install it. Will revert back as to whether this gets fixed. Thank you so much for your help!

pratyushmore, 2017-10-10:
@BoyuanChen @sytelus @sztomi 
Hello! The above solution works for linux but not Mac OS. I am planning to do a complete rebuild of AirSim. Is there any readme I could follow for Mac OS to ensure I am doing the right steps?

sytelus, 2017-10-10:
No, we haven't tried AirSim on MacOS. So that's a great area we would welcome any contributions!

pratyushmore, 2017-10-10:
@sytelus I feel I have been able to get through most things. However, I am now running into some problems with the build. I receive the following error:

ld: library not found for -lgcc_s
clang-3.9: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [output/bin/MavLinkTest] Error 1
make[1]: *** [MavLinkCom/MavLinkTest/CMakeFiles/MavLinkTest.dir/all] Error 2
make: *** [all] Error 2

Any ideas?

ProfFan, 2017-10-17:
@pratyushmore How did you manage to build the plugin? Mine is stuck at `struct nil { };` and I have no idea how to fix it. The above commit did not work for me.

pratyushmore, 2017-10-17:
Hi @ProfFan ,

I am instead using a Linux server. Installing as I type. Will update you.

ProfFan, 2017-10-18:
@pratyushmore Finally worked out everything. First you need to deal with `ld: library not found for -lgcc_s`, which you can just search in the source and delete `-lgcc -lgcc_s`. Second you deal with the `nil` issue by changing `rpclib` itself. Before upstream makes the change, you can see the corresponding commit in my [repo](http://github.com/ProfFan/rpclib/tree/dev). Be sure to change the files in `external/rpclib`. The code should now compile. If you encounter magic crashes, use my `settings.json` attached below.

After all that the code should run fine on macOS.

```
{ "SettingsVersion": 1, "SeeDocsAt": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md",
  "DefaultVehicleConfig": "Pixhawk",
  "SimMode": "",
  "ClockType": "",
  "ClockSpeed": 1,
  "LocalHostIp": "127.0.0.1",
  "RecordUIVisible": true,
  "LogMessagesVisible": true,
  "ViewMode": "",
  "UsageScenario": "",
  "RpcEnabled": true,
  "EngineSound": true,
  "PhysicsEngineName": "",
  "EnableCollisionPassthrogh": false,
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05
  },
  "CaptureSettings": [
    {
      "ImageType": -1,
      "Width": 256,
      "Height": 144,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100,
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ],
  "SubWindows": [
    {"WindowID": 0, "CameraID": 0, "ImageType": 3, "Visible": false},
    {"WindowID": 1, "CameraID": 0, "ImageType": 5, "Visible": false},
    {"WindowID": 2, "CameraID": 0, "ImageType": 0, "Visible": false}    
  ] ,
  "SimpleFlight": {
    "FirmwareName": "SimpleFlight",
    "DefaultVehicleState": "Armed",
    "RC": {
      "RemoteControlID": 0,
      "AllowAPIWhenDisconnected": false,
      "AllowAPIAlways": true
    },
    "ApiServerPort": 41451
  },
  "Pixhawk": {
    "FirmwareName": "PX4",
    "LogViewerHostIp": "127.0.0.1",
    "LogViewerPort": 14388,
    "OffboardCompID": 1,
    "OffboardSysID": 134,
    "SerialBaudRate": 115200,
    "SerialPort": "*",
    "SimCompID": 42,
    "SimSysID": 142,
    "UseSerial": false,
    "VehicleCompID": 1,
    "VehicleSysID": 135,
    "ApiServerPort": 41451
  }
}
```

alessandro-montanari, 2017-11-11:
I am having the same problem on El Capitan.
I deleted the flags `-lgcc -lgcc_s` and I have modified the file nil.hpp as described by @ProfFan. When I import the Blocks project in Unreal I don't get any popup and after some time it fails with this error:
```
[2/361] Compile Module.AirSim.gen.cpp
In file included from /Users/alessandro/Documents/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Mac/UE4Editor/Development/AirSim/Module.AirSim.cpp:32:
/Users/alessandro/Documents/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:125:10: fatal error: 'libudev.h' file not found
#include <libudev.h>
         ^
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /Users/alessandro/Documents/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Mac/UE4Editor-AirSim.dylib
Total build time: 149.46 seconds (Local executor: 0.00 seconds)
```
@ProfFan how did you fix this on mac?

ProfFan, 2017-11-14:
@alessandro-montanari  I think I simply deleted this line or installed some brew package for it. Been out now, will update you in a few days, after I go back to work.

sytelus, 2017-11-15:
I've made few changes that will hopefully make OSX build more easier.

1. Removed dependency of libudev header
2. Removed -lgcc and -gcc_s flags

MrChill, 2017-11-20:
Can you confirm that building Airsim on mac is working by just changing external/rpclib?
I tried but Blocks environment build is always failing

sytelus, 2018-02-13:
Several more fixes for Mac build was contributed recently:

- use latest rpclib
- add brew install coreutils
- use greadlink instead of readlink

Please try out the new version. Also, please file new issue so we can track it better.

closed 2018-02-13 by sytelus

Maddad21, 2022-12-26:
[image]