microsoft/AirSim #472: Step-by-Step HelloDrone

spencerh-b, 2017-09-19:
Is anyone able to help me just get the hellodrone working, trying to figure out this whole API thing, but I need somewhere to start. Just a quick list of what I need to do would be super helpful, thank you.

sytelus, 2017-09-19:
I'm wondering why you are confused? Is there any docs you read?

DroneServer is not required. It's only useful if you want to deploy your code on real vehicle.

DroneShell is not required for API access. It's just client shell-like program that uses APIs.

To use HelloDrone, just download binaries or build. Run the Blocks environment (press Play in UE editor if you are building yourself). Then start another instance of Visual Studio, set PythonClient project as startup project, set hello_drone.py as start script. Hit Run.

spencerh-b, 2017-09-19:
What would this process look like in linux?

closed 2017-09-20 by spencerh-b