microsoft/AirSim #473: Can't use RC control in latest version

jimchenhub, 2017-09-19:
I used to use old version Airsim (version about one month ago) in HITL way. When I change to the latest version which use `SimpleFlight` as default, I can't use RC to control the drone any more.

I use the default setting on [https://github.com/Microsoft/AirSim/blob/master/docs/settings.md](https://github.com/Microsoft/AirSim/blob/master/docs/settings.md) and change ImageType to be 0, Here is the log messages at start, it shows that `RC controller on USB not detected`.

[image]

Should I change some other setting to use RC?

sytelus, 2017-09-22:
simple_flight is software in loop so it requires RC with USB connectivity. See [this doc](https://github.com/Microsoft/AirSim/blob/master/docs/remote_control.md). For hardware in loop, you will need to use Pixhawk and connect RC directly to it.

jimchenhub, 2017-09-25:
I changed the 'DefaultVehicleConfig' to 'PX4', then everything is ok.
Thanks a lot.

closed 2017-09-25 by jimchenhub