microsoft/AirSim #475: Build own Executables with CMake - how setup the dependencies?

JonathanSchmalhofer, 2017-09-20:
I want to setup my own application using the AirSim-API. To Keep the codebase as clean as possible, I Keep my Code in a separate Folder. I also use CMake for Building (as does rpclib) for convenience, as I find it a bit easier to setup for an automated build Environment.

As a first start, I tried to build the "**HelloDrone**" Example in

    /path/to/myCode/src

instead of

    /path/to/AirSim/HelloDrone

This works so far, if I comment out all Code dependencies to AirSim, librpc and so on. But when Building the full example Code I of Course get Compiler Errors because of headers not being found.
I know that I could add all the dependencies manually - worst case hardcoded, but is there some way to include airsim and all additional dependencies using **find_package()**?

sytelus, 2017-11-14:
This is good idea but it effects very few of users. We would be glad to get pull request on this issue.

rakeshshrestha31, 2018-01-05:
Found a workaround by linking the necessary static .a libraries, doesn't use the **find_package()** though. Check this repo: [](https://github.com/rakeshshrestha31/airsim_hellodrone)

phildue, 2018-04-04:
This worked for me:
```
cmake_minimum_required(VERSION 3.5.0)
project(demo)
set(AIRSIM_ROOT ../../../AirSim)
LIST(APPEND CMAKE_MODULE_PATH "${AIRSIM_ROOT}/cmake/cmake-modules") 
INCLUDE("${AIRSIM_ROOT}/cmake/cmake-modules/CommonSetup.cmake")

CommonSetup()

add_subdirectory(${AIRSIM_ROOT}/external/rpclib/rpclib-2.2.1 ${CMAKE_CURRENT_BINARY_DIR}/rpclib)

CommonSetup()

IncludeEigen()

SetupConsoleBuild()

## Specify additional locations of header files
include_directories(
  ${AIRSIM_ROOT}/AirLib/include
  ${RPC_LIB_INCLUDES}
link_directories(${AIRSIM_ROOT}/AirLib/lib/x64/Debug/)

set(SOURCES
			src/main.cpp
			)
add_executable(${PROJECT_NAME} ${SOURCES})

CommonTargetLink()
target_link_libraries(${PROJECT_NAME} AirLib)
target_link_libraries(${PROJECT_NAME} ${RPC_LIB})
```

wardzacharyj, 2018-06-21:
For me I needed to specify Clang and libc++ be used

Also make sure you have installed the following: `sudo apt install libc++-dev`

```
cmake_minimum_required(VERSION 2.8.3)
project(some_project_name)

set(CMAKE_C_COMPILER /usr/bin/clang)
set(CMAKE_CXX_COMPILER /usr/bin/clang++)
set(CMAKE_CXX_FLAGS "${CMAKE_CXX_FLAGS} -stdlib=libc++")

LIST(APPEND CMAKE_MODULE_PATH ./AirSim/cmake/cmake-modules)
INCLUDE("./AirSim/cmake/cmake-modules/CommonSetup.cmake")
CommonSetup()
IncludeEigen()
SetupConsoleBuild()

message(${RPC_LIB_INCLUDES})

## Specify additional locations of header files
include_directories(
  ${PROJECT_SOURCE_DIR}/include
  ${AIRSIM_ROOT}/AirLib/include
  ${RPC_LIB_INCLUDES}
  ${AIRSIM_ROOT}/MavLinkCom/include
  ${AIRSIM_ROOT}/MavLinkCom/common_utils
)

## Declare a cpp executable
add_executable(${PROJECT_NAME} src/some_exe.cpp)

CommonTargetLink()
target_link_libraries(${PROJECT_NAME} ${AIRSIM_ROOT}/AirLib/lib/libAirLib.a )
target_link_libraries(${PROJECT_NAME} ${AIRSIM_ROOT}/AirLib/deps/rpclib/lib/librpc.a)
```

RVBldr, 2018-11-21:
Internal VSTF 7413.

closed 2020-08-13 by ironclownfish