microsoft/AirSim #479: Fly with QgroundControl

lyhbuaa, 2017-09-22:
I have implemented the simulation based on pixhawk，and I can fly with the RC，Now I want to fly it with QgroundControl，I want to edit the flight plan on the ground station, then send it to Pixhawk. The uav will fly based on my flight plan in the airsim simulator with Pixhawk HIL mode.

lyhbuaa, 2017-09-22:
Does anyone have realized it?

wifibot78, 2017-09-22:
Hello, 

We are doing this at this moment with my interns, it is working with firmware 1.5.x and not working well with 1.6.5 because of something wrong at init ? and we still have problem detecting landing, but I believe it is only a tuning of the landing detector.

Another thing to avoid disturbing the simulator on our test pc, we have wired a usb/rs232ttl cable to the telemetry port 1 and use qground on this serial port. we are able to setup a mission and run it.

PX4 HIL is crucial for our development so we hope solving 1.6.5 HIL problem.

Laurent

closed 2017-11-14 by sytelus