microsoft/AirSim #480: Not Building VS

spencerh-b, 2017-09-22:
```
1>------ Build started: Project: Blocks, Configuration: DebugGame_Game x64 ------
1>  Creating makefile for Blocks (no existing makefile)
1>  Performing full C++ include scan (no include cache file)
1>  Parsing headers for Blocks
1>    Running UnrealHeaderTool "C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\Blocks\DebugGame\Blocks.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>  Reflection code generated for Blocks in 6.6600012 seconds
1>  Performing 10 actions (4 in parallel)
1>  PCH.UELinkerFixups.cpp
1>  PCH.Blocks.cpp
1>  Module.AirSim.cpp
1>  AirSim.generated.cpp
1>  [5/10] Resource PCLaunch.rc
1>  UELinkerFixups.cpp
1>  [7/10] Resource PCLaunch.rc
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(536): warning C4005: 'TEXT': macro redefinition
1>  c:\program files\epic games\ue_4.16\engine\source\runtime\core\public\HAL/Platform.h(822): note: see previous definition of 'TEXT'
1>
1>  You have included windows.h before MinWindows.h
1>  All useless stuff from the windows headers won't be excluded !!!
1>
1>C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\controllers/MavLinkDroneController.hpp(7): fatal error C1083: Cannot open include file: 'MavLinkVehicle.hpp': No such file or directory
1>  Blocks.cpp
1>ERROR : UBT error : Failed to produce item: C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Binaries\Win64\UE4-AirSim-Win64-DebugGame.lib
1>  Total build time: 54.18 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" Blocks Win64 DebugGame "C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

I have gone through the instructions a couple times, still not able to figure out what is happening

spencerh-b, 2017-09-22:
Multiple repo's do not exist. For example, Droneshell appears to be missing build repo

sytelus, 2017-09-22:
Did you followed step 2 in this instructions: https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md

most likely Plugins\AirSim folder doesn't exist under Blocks project folder.

spencerh-b, 2017-09-25:
@sytelus The airsim folder exists, but it seems incomplete. I did indeed follow step 2

DavidLSmyth, 2017-09-26:
Also having the same issue, windows 10 VS 2015 with exact same build output. Plugins\AirSim folder contains Binaries, Content, Intermediate and Source folders, and a file called AirSim.uplugin.

sytelus, 2017-09-28:
This is very strange. Are you sure that you are double clicking on `Blocks\update_from_git.bat` file?

This batch file is very simple:
```
robocopy /MIR ..\..\Plugins\AirSim Plugins\AirSim /XD temp *. /njh /njs /ndl /np
robocopy /MIR ..\..\..\AirLib Plugins\AirSim\Source\AirLib /XD temp *. /njh /njs /ndl /np
cmd /c clean.bat
cmd /c GenerateProjectFiles.bat
```

The 2nd line copies AirLib folder which should have had missing file. Could you please check if `Plugins\AirSim\Source\AirLib` exists? Does it has files? If not, could you please try running this command from `Blocks` folder:
```
robocopy /MIR ..\..\..\AirLib Plugins\AirSim\Source\AirLib /XD temp *. /njh /njs /ndl /np
```

Please let me know. I'd really like to get to the bottom of this...

Mayankm96, 2017-09-28:
Well, I am facing the same issue currently. I checked the `AirSim\Unreal\Plugins\AirSim\Source` folder and the AirLib directory exists along with its contents.

sytelus, 2017-09-28:
Ok... here are some steps for debugging. Could you confirm result of following steps?

1. First make sure you have run `build` command in AirSim folder from VS x64 command prompt and that build was successful. You should have this file exist: `AirSim\AirLib\deps\MavLinkCom\include\MavLinkVehicle.hpp`.

2. Go to `Unreal\Environments\Blocks` project. Double click on `update_from_git.bat`. After this step you should have this file exist: `AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\include\MavLinkVehicle.hpp`

3. Double click on .sln file. Make sure you have `DebugGame-Editor` and `x64` build configuration selected. Hit F5.

Mayankm96, 2017-09-28:
I started the build process again (by first running `clean.bat` and then doing `git clean -dxf`. Step 1 and 2 are cleared since the files you have mentioned are present over at their respective places.  Step 3, however is leading to the same error as before:
```
>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
1>  Creating makefile for BlocksEditor (no existing makefile)
1>  Performing full C++ include scan (no include cache file)
1>  Parsing headers for BlocksEditor
1>    Running UnrealHeaderTool "C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\BlocksEditor\DebugGame\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>  Reflection code generated for BlocksEditor in 6.9423431 seconds
1>  Performing 11 actions (4 in parallel)
1>  [2/11] Resource ModuleVersionResource.rc.inl
1>  [3/11] Resource PCLaunch.rc
1>  [4/11] Resource PCLaunch.rc
1>  SharedPCH.Engine.NonOptimized.cpp
1>  Module.AirSim.cpp
1>  Blocks.cpp
1>  AirSim.generated.cpp
1>  [8/11] Link UE4Editor-Blocks-Win64-DebugGame.lib
1>     Creating library C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Blocks-Win64-DebugGame.lib and object C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Blocks-Win64-DebugGame.exp
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(536): warning C4005: 'TEXT': macro redefinition
1>  c:\program files\epic games\ue_4.16\engine\source\runtime\core\public\HAL/Platform.h(822): note: see previous definition of 'TEXT'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(17005): warning C4668: '_APISET_RTLSUPPORT_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\interlockedapi.h(73): warning C4668: '_APISET_INTERLOCKED_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\processthreadsapi.h(1170): warning C4668: '_WIN32_WINNT_WINTHRESHOLD' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(357): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(472): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\securitybaseapi.h(1302): warning C4668: '_APISET_SECURITYBASE_VER' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\winbase.h(8618): warning C4668: 'NTDDI_WIN7SP1' is not defined as a preprocessor macro, replacing with '0' for '#if/#elif'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\controllers/Waiter.hpp(24): error C4003: not enough actual parameters for macro 'max'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\controllers/Waiter.hpp(24): error C2589: '(': illegal token on right side of '::'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\include\controllers/Waiter.hpp(24): error C2059: syntax error: '::'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(23): error C2589: '(': illegal token on right side of '::'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(23): error C2059: syntax error: '::'
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(159): error C2664: 'msr::airlib::Waiter::Waiter(msr::airlib::Waiter &&)': cannot convert argument 1 from 'float' to 'const msr::airlib::Waiter &'
1>  C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(159): note: Reason: cannot convert from 'float' to 'const msr::airlib::Waiter'
1>  C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(159): note: No constructor could take the source type, or constructor overload resolution was ambiguous
1>C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(299): error C2664: 'msr::airlib::Waiter::Waiter(msr::airlib::Waiter &&)': cannot convert argument 1 from 'float' to 'const msr::airlib::Waiter &'
1>  C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\src\controllers\DroneControllerBase.cpp(299): note: Reason: cannot convert from 'float' to 'const msr::airlib::Waiter'
[... cut ...]
1>  Total build time: 24.27 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 DebugGame "C:\Users\BERSERK\Desktop\Mayank\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========
```

sytelus, 2017-09-28:
I just fixed above issue. Would you like to try again? Thanks for helping us debug this issue.

spencerh-b, 2017-10-03:
@sytelus 
`1>------ Build started: Project: Blocks, Configuration: DebugGame_Editor x64 ------
1>  Performing full C++ include scan (building a new target)
1>  Creating makefile for BlocksEditor (no existing makefile)
1>  Parsing headers for BlocksEditor
1>    Running UnrealHeaderTool "C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" "C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\BlocksEditor\DebugGame\BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors -installed
1>  Reflection code generated for BlocksEditor in 8.8460935 seconds
1>  Performing 12 actions (4 in parallel)
1>  [2/12] Resource ModuleVersionResource.rc.inl
1>  [4/12] Resource PCLaunch.rc
1>  SharedPCH.Engine.NonOptimized.cpp
1>  PCH.Blocks.cpp
1>  [5/12] Resource PCLaunch.rc
1>  Blocks.cpp
1>  [7/12] Link UE4Editor-Blocks-Win64-DebugGame.lib
1>     Creating library C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Blocks-Win64-DebugGame.lib and object C:\AirSim\Unreal\Environments\Blocks\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-Blocks-Win64-DebugGame.exp
1>  Module.AirSim.cpp
1>  AirSim.generated.cpp
1>C:\Program Files (x86)\Windows Kits\8.1\include\um\winnt.h(536): warning C4005: 'TEXT': macro redefinition
1>  c:\program files\epic games\ue_4.16\engine\source\runtime\core\public\HAL/Platform.h(822): note: see previous definition of 'TEXT'
1>
1>  You have included windows.h before MinWindows.h
1>  All useless stuff from the windows headers won't be excluded !!!
1>
1>  [10/12] Link UE4Editor-AirSim-Win64-DebugGame.lib
1>  [11/12] Link UE4Editor-AirSim-Win64-DebugGame.dll
1>     Creating library C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-Win64-DebugGame.lib and object C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Intermediate\Build\Win64\UE4Editor\DebugGame\UE4Editor-AirSim-Win64-DebugGame.exp
1>  [12/12] Link UE4Editor-Blocks-Win64-DebugGame.dll
1>LINK : fatal error LNK1181: cannot open input file 'C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib'
1>LINK : fatal error LNK1181: cannot open input file 'C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib'
1>ERROR : UBT error : Failed to produce item: C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Binaries\Win64\UE4Editor-AirSim-Win64-DebugGame.dll
1>  Total build time: 151.98 seconds (Local executor: 0.00 seconds)
1>C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets(41,5): error MSB3075: The command ""C:\Program Files\Epic Games\UE_4.16\Engine\Build\BatchFiles\Build.bat" BlocksEditor Win64 DebugGame "C:\AirSim\Unreal\Environments\Blocks\Blocks.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.
========== Build: 0 succeeded, 1 failed, 0 up-to-date, 0 skipped ==========`

It progressed much further, but still failing

sytelus, 2017-10-04:
Please look at the error and see if you can help debug.  

1. Does this file exist? `C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib`?

2. If not then does it exist in `AirSim\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib`?

3. If not then did you do the build? If yes, then did you clicked on `update_from_git.bat`?

spencerh-b, 2017-10-06:
@sytelus After AirLib\deps the MavLinkCom folder does not exist

sytelus, 2017-10-06:
That means you haven't done build. Try running build.cmd. It should copy those files to deps. Then when you double click on update_from_git.bat, it will copy deps to Blocks project.

spencerh-b, 2017-10-06:
I have gone through the steps multiple times now and the MavLinkCom folder still does not appear. @sytelus

sytelus, 2017-10-07:
Do you mean you type build at VS x64 Command Prompt in AirSim folder and deps\MavLinkCom still doesn't exist?

spencerh-b, 2017-10-09:
@sytelus Thank you for the help, everything is running great now

closed 2017-10-09 by spencerh-b

Jackyjae, 2017-12-07:
@spencerh-b Hi, I met the same problem as you had and I dont know how to fix it at all. Cant finish build because of this: 
LINK : fatal error LNK1181: cannot open input file 'rpc.lib' [D:\Jack\AirSim-1.1.7\AirLib\AirLib.vcxproj]

spencerh-b, 2017-12-07:
@Jackyjae just make sure that you are running VS 2015 and insure that you use the VS x64 Command Line to complete all of the tasks, don't just click on the build.cmd file in the folder, that will not complete the build.

Jackyjae, 2017-12-14:
@spencerh-b sorry to border you again~ Did you use the APIs of AieSim? I tried the sample of using "getOneImage" but it can't work and give me the error "cannot open include file "vehicles/multirotor/api/MultirotorRpcLibClient.hpp" ". I tried to add include and library directories in the property page of vs and it's still not working. BTW, it also has another error like this: 
`Severity	Code	Description	Project	File	Line	Suppression State
Error	MSB3075	The command ""D:\Epic Games\UE_4.18\Engine\Build\BatchFiles\Build.bat" Task1Editor Win64 DebugGame "C:\Users\angus\Documents\Unreal Projects\Task1\Task1.uproject" -waitmutex" exited with code 5. Please verify that you have sufficient rights to run this command.	Task1	C:\Program Files (x86)\MSBuild\Microsoft.Cpp\v4.0\V140\Microsoft.MakeFile.Targets	41	
`
Could you help me to fix it if you know? Thanks so much.

spencerh-b, 2018-01-30:
@Jackyjae I'm sorry for responding so late I have been on vacation. It sounds like the build command didnt execute properly make sure you run it as an admin from with in the VS Native command prompt. I have as well experienced this issue and this seemed to fix my problems

Isaac45, 2018-05-07:
I can't build airsim, using vs 2017. Please help resolve this error

"D:\AirSim\AirSim.sln" (default target) (1) ->
"D:\AirSim\DroneShell\DroneShell.vcxproj" (default target) (2) ->
"D:\AirSim\AirLib\AirLib.vcxproj" (default target) (3) ->
"D:\AirSim\MavLinkCom\MavLinkCom.vcxproj" (default target) (4) ->
(ClCompile target) ->
  c:\program files (x86)\windows kits\8.1\include\um\shlobj.h(1054): error C2220: warning treated as error - no 'object
' file generated [D:\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\airsim\mavlinkcom\src\serial_com\tcpclientport.cpp(19): fatal error C1083: Cannot open include file: 'winsock2.h':
 No such file or directory [D:\AirSim\MavLinkCom\MavLinkCom.vcxproj]
  d:\airsim\mavlinkcom\src\serial_com\udpclientport.cpp(19): fatal error C1083: Cannot open include file: 'winsock2.h':
 No such file or directory [D:\AirSim\MavLinkCom\MavLinkCom.vcxproj]

    1 Warning(s)
    3 Error(s)

shad7867, 2018-09-16:
Hi Isaac45, was the above issue resolved? I am at the same place, can you advise solution if you got past it, or can anyone else advise.

thanks you much in advance, Shad7867

racheraven, 2021-08-15:
@sytelus 

> Please look at the error and see if you can help debug.
> 
> 1. Does this file exist? `C:\AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib`?
> 2. If not then does it exist in `AirSim\AirLib\deps\MavLinkCom\lib\x64\Release\MavLinkCom.lib`?
> 3. If not then did you ran the build.cmd? If yes, then did you clicked on `update_from_git.bat` in Blocks folder?

What should I do if I fall into the 2nd kind of error when packaging a project for Linux on a Windows computer?
I am currently using a 4.25v UE4 and the latest version (v1.5.0) of airsim.

Vampireqyz, 2022-01-13:
> 好的...这里有一些调试步骤。您能否确认以下步骤的结果？
> 
> 1. 首先确保您`build`已从 VS x64 命令提示符运行AirSim 文件夹中的命令，并且构建成功。你应该有这个文件存在：`AirSim\AirLib\deps\MavLinkCom\include\MavLinkVehicle.hpp`.
> 2. 去`Unreal\Environments\Blocks`项目。双击`update_from_git.bat`。在这一步之后，你应该有这个文件存在：`AirSim\Unreal\Environments\Blocks\Plugins\AirSim\Source\AirLib\deps\MavLinkCom\include\MavLinkVehicle.hpp`
> 3. 双击 .sln 文件。确保您已选择`DebugGame-Editor`并`x64`构建配置。按 F5。

we have the problem when we double click".sln"file.and choose debuggame editor and win64,f5
but it failed,there is a problem with the following picture.
[image]
[image]
,