microsoft/AirSim #481: LandedState Bug on Python

RodgerLuo, 2017-09-22:
On PythonClient.py, class LandedState shows that 0 indicates Landed and 1 indicates Flying. 

[image]

However, I asked the drone to send a landed state immediately after the takeoff behavior is accomplished, and it returned 0, whereas to my understanding it should return 1.

[image]
 
[image]

Please let me know if it's a bug or I did something wrong.

sytelus, 2017-11-14:
@lovettchris might be interested in this.

lovettchris, 2017-12-09:
I'm seeing the correct behavior here now that this [pull request ](https://github.com/Microsoft/AirSim/pull/670) is merged.  The problem was that the drone thought it was in the air when it should have been on the ground.

closed 2017-12-09 by lovettchris