microsoft/AirSim #486: AirSim_FrSkyTaranis.bin fails to be installed on FrSky Taranis X9D Plus

PanGroup, 2017-09-26:
Following Note on FrSky Taranis X9D Plus:
"FrSky Taranis X9D Plus is real UAV remote control with an advantage that it has USB port so it can be directly connected to PC. You can download AirSim config file and follow this tutorial to import it in your RC. You should then see "sim" model in RC with all channels configured properly."

I have download the AirSim config file from https://github.com/Microsoft/AirSim/blob/master/docs/misc/AirSim_FrSkyTaranis.bin, which is then copied to the models folder of the X9D Plus sd card. However, when I am trying to restore "airsim_frskytaranis" model using the X9D Plus, it fails to install with an error "Incompatible". I have updated the firmware to the latest version, but this error still remains.

PanGroup, 2017-09-28:
The problem has been solved. AirSim_FrSkyTaranis.bin file downloaded from https://github.com/Microsoft/AirSim/blob/master/docs/misc/AirSim_FrSkyTaranis.bin via webpage is not exactly the one for Taranis X9D Plus. The correct file should be downloaded via github client.

sytelus, 2017-09-28:
I'm assuming you had download trouble from GitHub? I think downloading "raw" data should work.

closed 2017-09-28 by sytelus

AloshkaD, 2018-06-17:
@PanGroup I've downloaded it from github and still getting incompatible. What opentx version do you have?

AloshkaD, 2018-06-27:
I've created a new config file

Ahmad-Drak, 2019-04-17:
@AloshkaD not working on my end. did your bin file work?

zardosht, 2022-02-02:
@AloshkaD Can you please share your config file?