microsoft/AirSim #487: Getting Acceleration

hngenc, 2017-09-26:
Is there an API call to get the drone's acceleration? I'd like something like the `getVelocity()` function in `RpcLibClient.hpp`

The main reason I want this is because I'd like to populate [ROS IMU messages](http://docs.ros.org/api/sensor_msgs/html/msg/Imu.html). I can populate the velocity and orientation fields myself, but not the acceleration fields.

Ricardus312, 2017-09-27:
Maybe I can help you if you provide more information:
- Which acceleration do you want? Ground truth from the physics engine or simulated estimations from the IMU?
- What is your setup? Software in the loop or hardware in the loop?

hngenc, 2017-09-27:
Thanks for your response, @Ricardus312!

- I'd like simulated estimations from the IMU.
- I can use either SITL or HITL, although I'd prefer a solution that works with the simple_flight_controller.

Ricardus312, 2017-09-28:
Well, it seems we both have the same problem (https://github.com/Microsoft/AirSim/issues/476). I haven't figured out how to get the acceleration from the IMU yet, but I think I am pretty close from it. I will share with you all what I have discovered until now, maybe it could be useful for you:

- SitL setup (Simple Flight). On simple flight, if you press 'R' button during simulation, it starts to record flight images and writing a log file. That log file outputs the position and orientation of the drone, sampling several times per second, and you can modify easily the code of the log function to make it output velocities and accelerations, both linear and angular. Here is how to get velocities and accelerations in simple flight:
    - Go to the directory of your Unreal project (usually at \Documents\Unreal Projects\Project_name\ in Windows, but if you are using the default Blocks environment that comes with AirSim, it could be on \AirSim\Unreal\Environments\Blocks\)
    - Open the 'Project.sln' file with Visual Studio (e.g. if you are using the default blocks environment, the file is Blocks.sln).
    - On the solutions explorator, navigate to \Games\Project\Plugins\AirSim\Source\Recording\RecordingFile.cpp" and open it (other option is going directly to the directory "..\Project_name\Plugins\AirSim\Source\Recording\" and open with Visual Studio the file "RecordingFile.cpp".
    - About line 50 of that file, you will see this code:
    line.append(std::to_string(timestamp_millis)).append("\t")
        .append(std::to_string(kinematics.pose.position.x())).append("\t")
        .append(std::to_string(kinematics.pose.position.y())).append("\t")
        .append(std::to_string(kinematics.pose.position.z())).append("\t")
        .append(std::to_string(kinematics.pose.orientation.w())).append("\t")
        .append(std::to_string(kinematics.pose.orientation.x())).append("\t")
        .append(std::to_string(kinematics.pose.orientation.y())).append("\t")
        .append(std::to_string(kinematics.pose.orientation.z())).append("\t")

      All you have to do is add to that code these lines, and you will get all the velocities and accelerations 
      (ground truth):
                .append(std::to_string(kinematics.twist.linear.x())).append("\t")
		.append(std::to_string(kinematics.twist.linear.y())).append("\t")
		.append(std::to_string(kinematics.twist.linear.z())).append("\t")
		.append(std::to_string(kinematics.twist.angular.x())).append("\t")
		.append(std::to_string(kinematics.twist.angular.y())).append("\t")
		.append(std::to_string(kinematics.twist.angular.z())).append("\t")
		.append(std::to_string(kinematics.accelerations.linear.x())).append("\t")
		.append(std::to_string(kinematics.accelerations.linear.y())).append("\t")
		.append(std::to_string(kinematics.accelerations.linear.z())).append("\t")
		.append(std::to_string(kinematics.accelerations.angular.x())).append("\t")
		.append(std::to_string(kinematics.accelerations.angular.y())).append("\t")
		.append(std::to_string(kinematics.accelerations.angular.z())).append("\t")

    - Also, if you don't want to get the screen captures, you only have to replace (or comment) the line 21 
    of that file:
    imageSavedOk = FFileHelper::SaveArrayToFile(image_data, *filePath);
    For this other one: 
    imageSavedOk = true;

    - Regarding the IMU accelerations, if you read the paper 
    (https://github.com/Microsoft/AirSim/blob/master/docs/paper/main.pdf), in the point 3.4 it says that 
    > "AirSim offers sensor models for accelerometer, gyroscope, barometer, magnetometer
    and GPS. All our sensor models are implemented as C++ header-only library and
    can be independently used outside of AirSim"  
    
    So in theory, it is possible to write a code that just imports the IMU sensor, calls a method and gets 
    the angular velocities and accelerations from it. Unfortunately, I am not a C++ programmer (yet) and 
    my understanding of the code is very limited, so I still haven't been able to do this by myself. The code 
    for the sensors is at "..\Project_name\Plugins\AirSim\Source\AirLib\include\sensors\". Check out the 
    code in "..\imu\ImuSimple.hpp". I think that somebody with good C++ skills could easily modify the 
    code in "RecordingFile.cpp" to call the IMU ang get its outputs, but I still haven't that "good C++ 
    skills".

- For the Hardware int the Loop setting, theoretically it is possible to get the MavLink messages between your external flight controller hardware and the simulator using an external program called Px4LogViewer, look at the docs (https://github.com/Microsoft/AirSim/blob/master/docs/px4_logging.md).

These are all my insights until now, I hope this maybe useful for you. If you discover before me how to get that IMU data, please share with me!!!

Regards

abdelrahman-t, 2017-12-09:
reporter.writeValue("Lin-Vel", current_.twist.linear);
reporter.writeValue("Lin-Accl", current_.accelerations.linear);
reporter.writeValue("Ang-Vel", current_.twist.angular);
reporter.writeValue("Ang-Accl", current_.accelerations.angular);

could you please verify in which frame are these measurements in?

sytelus, 2017-12-16:
We have added new APIs to get angular velocity (along with other kinematics quantities). Please see https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#getmultirotorstate.

For cars, this is available in [getCarState API](https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#apis-for-car).

Note that above are ground truth quantities and you should not use it as sensor data. We'll be adding sensor APIs as well soon.

closed 2018-02-13 by sytelus

Fab7c4, 2018-03-04:
Hey @sytelus ,
are there any news regarding sensor API's? I could not find a specific issue for that yet.

DrifterFun, 2018-11-13:
Is there any updating about add adding sensor APIs (For IMU & GPS etc)?