microsoft/AirSim #488: Working Project?

spencerh-b, 2017-09-26:
Does anyone have a working project on linux they would be willing to throw on onedrive?

rjk2147, 2017-09-27:
Did you try running the install_run_all.sh script instead of following the directions here: https://github.com/Microsoft/AirSim/blob/master/docs/build_linux.md 

The script worked for me but the directions didnt, probably unreal 4.17 that works better

sytelus, 2017-09-28:
Could you please post the error. I was able to build it all fine.

spencerh-b, 2017-10-01:
'spencer@exalabs-OptiPlex-9020:~/AirSim/HelloDrone$ gpp main.cpp
// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

main.cpp:5: error: Requested include file not found'

spencerh-b, 2017-10-01:
`_____@e______-OptiPlex-9020:~/AirSim/HelloDrone$ gpp main.cpp
// Copyright (c) Microsoft Corporation. All rights reserved.
// Licensed under the MIT License.

main.cpp:5: error: Requested include file not found`

closed 2017-10-03 by spencerh-b

SaiVineethJ, 2018-02-02:
Did you find a way to remove the error of include file missing??

spencerh-b, 2018-02-02:
Instead of using gpp you use clang for AirSim, I opted to just move to the python version in the python client folder, it could also mean that your repo didn't build right but I would just try.

```
cd <AirSim Repo>
python PythonClient/hello_drone.py
```

Hopefully this works for you @SaiVineethJ

SaiVineethJ, 2018-02-09:
Thanks it worked for me..

But the drone is not moving on running the hello_drone.py, the hello_car.py is working properly.
Is your drone showing any changes on running the code?? 
I am using Ubuntu14.

SaiVineethJ, 2018-02-20:
Thanks it worked. I had to use default settings of AirSim initially.