microsoft/AirSim #489: how to reset environment?

akshaysw, 2017-09-27:
how to reset position of drone while simulation is running and start a new episode just like reset function from OpenAI gym ''env.reset()". @sytelus  [#331](https://github.com/Microsoft/AirSim/issues/331)

Kjell-K, 2017-09-27:
Basically simSetPose() does the trick. There is an [issue though](https://github.com/Microsoft/AirSim/issues/474).

I am using the PythonClient
Example code:

```
client = AirSimClient('127.0.0.1')
client.confirmConnection()
client.enableApiControl(True)
client.armDisarm(True)
pos = client.getPosition()
ori = client.getOrientation()

while crash_count < number_episodes:

    client.simSetPose(Vector3r(pos.x_val, pos.y_val, pos.z_val),Quaternionr(ori.w_val, ori.x_val, ori.y_val, ori.z_val))
        
    client.armDisarm(True)    
    client.moveToZ(z, 3)  
.
.
.
```

sytelus, 2017-09-28:
We still need to add API for reset. Unfortunately that warrents some refactoring so its taking time...

Kjell-K, 2017-09-28:
@sytelus So what is the simSetPose() supposed to do then?
I have trouble with that one though. See [issue.](https://github.com/Microsoft/AirSim/issues/474)

Kjell-K, 2017-09-28:
Workaround SOLVED:
I want to go to my home position from any position where I crashed right now. 

My Workaround due to:
1. simSetPose() takes direct and physical path --> collides with obstacles

- Since I am working with an environment with open sky, I escape to the top (-30) and move from there

2. simSetPose() is very unreliable in actually performing the reset

- I have to do multiple attempts until the copter  really sets the position. There fore the while and step by step if loops. time.sleep(0.1) is important

```
home_pos = client.getPosition()

def reset_env():
    
    reset = False

    while reset != True:
        now = client.getPosition()
        client.simSetPose(Vector3r(now.x_val, now.y_val, -30),Quaternionr(home_ori.w_val, home_ori.x_val, home_ori.y_val, home_ori.z_val))
        time.sleep(0.1)
        now = client.getPosition()
        print(now.z_val)
        
        if abs(now.z_val - (-30)) < 1:
            client.simSetPose(Vector3r(home_pos.x_val, home_pos.y_val, -30),Quaternionr(home_ori.w_val, home_ori.x_val, home_ori.y_val, home_ori.z_val))
            time.sleep(0.1)
            now = client.getPosition()
            print(now.x_val, now.y_val, now.z_val)
            
            if abs(now.x_val - home_pos.x_val) < 1 and abs(now.y_val - home_pos.y_val) < 1 and abs(now.z_val - (-30)) < 1:
                client.simSetPose(Vector3r(home_pos.x_val, home_pos.y_val, home_pos.z_val),Quaternionr(home_ori.w_val, home_ori.x_val, home_ori.y_val, home_ori.z_val))
                time.sleep(0.1)
                now = client.getPosition()
                print(now.x_val, now.y_val, now.z_val)
                
                if abs(now.x_val - home_pos.x_val) < 1 and abs(now.y_val - home_pos.y_val) < 1 and abs(now.z_val - home_pos.z_val) < 1:
                    reset = True
```

sytelus, 2017-10-04:
FYI - I've implemented `reset` API in car. I'm planning to add it to multirotor as well but that would be coming later.

closed 2017-10-04 by sytelus

Kjell-K, 2017-10-06:
@sytelus I tested the  new simSetPose() in my test environment. After I run some episodes, it shows that the ignore_collison feature does not work properly. 

```
home = client.simGetPose()
.
.
client.simSetPose(home, True);
```

Here the attempt of a reset with simSetPose():

[image]

The copter still crashes on obstacles in direct path. Reopen or open a new issue?

cangokalp, 2017-10-12:
Hi! @Kjell-K first of all thank you for making this repository that interacts with the sim like an open ai gym environment. I've been testing it - when I run that code you suggested as a workaround I get 

"msgpackrpc.error.RPCError: b"rpclib: function 'simSetPose' (called with 2 arg(s)) threw an exception. The exception contained this information: bad cast."

cangokalp, 2017-10-12:
and then when I use it with one argument -say home coordiantes -  it would just start from beginning and go to the last location. Regardless of what I give to simSetPose - it takes me from beginning to where I was at before... simGetPose() does the same thing too. I don't think they are working properly

cangokalp, 2017-10-12:
@sytelus is the reset API to multirotor coming soon?

Kjell-K, 2017-10-12:
@cangokalp My posted code here is not longer valid after some updates. Refer to the new issue #532