microsoft/AirSim #494: Unable to run AirSim in Car mode

kaihuchen, 2017-09-28:
FYI, I have tried to run  AirSim in Car mode without success. I have changed settings.json per the instructions in https://github.com/Microsoft/AirSim/blob/master/docs/using_car.md , but AirSim is still showing a drone after it is restarted.  Running hello_car.py also gives me *Connection refused* message.

I am running the Windows version of AirSim, updated from the master branch just hours ago. My installation runs hello_drone.py perfectly without problem.

sytelus, 2017-09-28:
Are you using binaries? The binary release isn't updated yet...

kaihuchen, 2017-09-28:
Ah, I see. Indeed I am using binaries. Thanks for your quick reply!

sytelus, 2017-09-28:
Just uploaded new [release](https://github.com/Microsoft/AirSim/releases/tag/v1.1%2C0).

closed 2017-09-28 by sytelus

kaihuchen, 2017-09-29:
@sytelus    Works great! Thanks!