microsoft/AirSim #498: Car's roll bar showing up in the image

kaihuchen, 2017-09-29:
Having fun with the newly released Car model. One thing that I noticed is that part of the car's roll bar is permanently showing in the camera image, like the following (using a depth image as an example):

[image]

Wonder if it is possible to do something to avoid showing the roll bar in the camera images, perhaps by moving the camera's position forward a little, or by altering the car's structure (whichever that is easier).

kaihuchen, 2017-09-30:
To clarify: I am trying to use the depth information for the purpose of autonomous navigation, and the roll bars are interfering with the car's peripheral vision, and makes it hard for the car to find alternative routes.

sytelus, 2017-10-04:
Fixed: https://github.com/Microsoft/AirSim/commit/b76bbfa7fb3b4a32a095698be1cd6bd7eee15597

I also added driver's view and rear camera. You can try out following in settings.json to see all 3 cameras in action!

```
  "SubWindows": [
    {"WindowID": 0, "ImageType": 0, "CameraID": 3},
    {"WindowID": 1, "ImageType": 3, "CameraID": 0},
    {"WindowID": 2, "ImageType": 6, "CameraID": 4}
  ]
```

kaihuchen, 2017-10-05:
Works great! The newly added rear camera is really nice, since now the car is able to see what's behind it when trying to back up.

One minor thing though: the antenna visible in the rear camera while visually very interesting, it is however going to be an impediment for autonomous navigation (say, when the car is trying to go in reverse navigated by vision).

sytelus, 2017-10-06:
For this particular car, antenna leans too back and don't want to attach camera outside body. I hope to solve this in new car mesh we are planning.

sytelus, 2017-11-14:
FYI - the new car mesh is out in new release and looks much better. In addition we have now 5 camera on the car: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#available-cameras

In addition, we have now two types of depth images: planner and perspective: https://github.com/Microsoft/AirSim/blob/master/docs/image_apis.md#what-does-pixel-values-mean-in-different-image-types

closed 2017-11-14 by sytelus