microsoft/AirSim #500: Depth map got messed up under some circumstances

kaihuchen, 2017-09-30:
I am seeing incorrect reported depth information under some circumstances. This applies to both the DepthMeters mode as well as the new DepthPerspective mode. Below I will describe it using information collected with the older DepthMeters mode.

I have an autonomous car that navigates strictly based on depth information. In the beginning of a testing session the depth information always  looks fine. The car is also able to navigate around other parked cars on the street using the depth information without problem. However, whenever the car reaches the end of a T intersection and begins to backup, then the depth map got largely blanked out with uniformly incorrect distance information (i.e., see the GIF below, look for the large red area in the depth map as the car is backing up).

[image]

The above is perfectly reproducible. I have also seen this occur when the car goes off road (say, going into backyards).

Kjell-K, 2017-09-30:
Please refer to issue #491 and #478. There is a change in the master branch now with using DepthPlanner instead of DepthMeters. I have not tried it yet, but it is supposed to solve this issue as well.

kaihuchen, 2017-09-30:
FYI, my test above was already using the latest master branch from this morning.
There are some problems with DepthPlanner, where I got a strangely patterned depth map.

[image]

DepthPerspective works better than DepthPlanner, but got more or less the same problem as mentioned above.  You can see this in the screenshot below, where the gray rectangle in the upper-right corner shows what I got from DepthPerspective (converted froma 2D array of floats to RGB-gray).  There most of its left side is showing incorrect depth. 

[image]

As a reference, the depth map at the upper-right corner below shows what I got from a *normal* DepthPerspective map.

[image]

sytelus, 2017-10-01:
I have been looking at this issue. It seems that the problem is in this specific environment and its triggering some bug in Unreal. In Blocks environment things are working all fine. I'm wondering if there is any issue with sky light or some rendering settings in this environment. If you are willing to mess with Unreal, you are welcome to debug this as well :)

Mayankm96, 2017-10-01:
Hm well I am facing a bit of an issue with depth map too right now. I changed the PIP camera by 90 degrees so that they face towards the ground now instead of front, and the depth map doesn't seem correct. Any suggestions on this?

[image]

Kjell-K, 2017-10-01:
@sytelus For Blocks and Neighbourhood the issue is remaining with the new master.

[image]

sytelus, 2017-10-03:
Ah! Finally found the cause and have fixed it. Please try out the latest code. I'll update the binaries soon.

Kjell-K, 2017-10-03:
@sytelus Yes the 180° is solved! But on the other side the issue #491 as black/white over all the 360°. I will post there.

sytelus, 2017-10-04:
Binaries are updated. We also have new Downtown environment.

closed 2017-10-04 by sytelus

kaihuchen, 2017-10-04:
@sytelus    I am still seeing binaries posted from 6 days ago here [here](https://github.com/Microsoft/AirSim/releases/tag/v1.1.1). Am I looking at the wrong place?