microsoft/AirSim #501: getting angular velocities through api

abdelrahman-t, 2017-09-30:
is there anyway to get angular velocities through provided apis ?

Ricardus312, 2017-10-02:
Please, see this thread: https://github.com/Microsoft/AirSim/issues/487

abdelrahman-t, 2017-10-02:
I have made a fork, with added access to angular velocity through the python API, if anyone is interested

jpreiss, 2017-11-01:
@abdelrahman-t is the implementation in your fork tested/working? I also need this feature. Why don't you submit a pull request to mainline?

cangokalp, 2017-11-22:
I need this too! - can you guys please make angular velocity exposed in the python API. I actually did it once for an old version- but now some of the interactions has changed in the new version, can't get it to work anymore.

sytelus, 2017-12-16:
We have added new APIs to get angular velocity (along with other kinematics quantities). Please see https://github.com/Microsoft/AirSim/blob/master/docs/apis.md#getmultirotorstate.

closed 2018-02-13 by sytelus