microsoft/AirSim #502: AirSim does not build on Fedora

aerosuraj, 2017-10-01:
Currently setup.sh only works with Ubuntu. The scripts are not generalized for Fedora

sytelus, 2017-10-01:
Could you please post build log/error messages etc? We don't use Fedora so it would be great if you can help/test on Fedora :).

aerosuraj, 2017-10-01:
The issue is with setup.sh script in AirSim. It uses apt-get to install packages wheres in Fedora the command is dnf. I tried to find the dependencies that need to be installed but still had issues. Can you post a list of dependencies for AirSim? Also, need minimum system requirements for running AirSim. It seems my Linux machine with Ubuntu is not adequate either.

sytelus, 2017-10-03:
Main dependency is cmake and clang 3.9 which is what setup.sh tries to install. If you replace apt-get for those, things should work, I guess.

sytelus, 2017-11-14:
We would love to get any pull request for doc on how to build AirSim on Fedora :)

closed 2017-11-14 by sytelus