microsoft/AirSim #506: Can I use a car and a multirotor at the same time?

luchadorMalloc, 2017-10-02:
I wan to control a car using the image from a drone multirotor but how can I spawn one of different tyoes of robots

sytelus, 2017-10-03:
Currently no but this in our mind (specifically because of some challenges involves this scenario). We hope to enable this in near future.

frontw, 2018-01-26:
Hi! Any update on it?

sytelus, 2018-01-30:
Not yet. I don't anticipating this getting implemented in at least next 3 months. I would encourage to study the code and see if you can contribute.

DavidLSmyth, 2018-01-30:
> Currently no but this in our mind (specifically because of some challenges involves this scenario). We hope to enable this in near future.

Could you quickly outline some of the challenges mentioned @stylus? Also could you list the files you think would need to be changed to implement this? I'd like to try and contribute to this

kshitiz38, 2018-03-08:
Hi! I was able to achieve this, you can check my fork. For my scenario I wanted to be able to control the multirotor using APIs to track the car which I would drive around. I had to make modifications to SimModeWorldMultiRotor.cpp based on SimModeCar.cpp. So now when you start the simulation, select no to start multirotor mode and a car and drone both will be spawned.

kshitiz38, 2018-06-06:
@alvikag Fixed it..Check the latest commit.

Sorry, I had only tested the build for linux. However this issue is due to mkdir_p (recursive mkdir) in Filesystem.cpp, the function was supposed to be invoked for linux platform only. I have added the conditional compilation flags so it will hopefully build without any errors now!

alvikag, 2018-06-06:
Hi,
Thanks! I checked the latest commit, most of the errors are gone but the build is still failing.
[image]

alvikag, 2018-06-06:
As of now, I disabled the warning and now its getting compiled. Thanks!

kshitiz38, 2018-06-07:
@aadi-mishra...the name of simmode is just "Both" and if you keep it blank you will get 2 prompts...choose quadrotor simulation (No) in first and when second appears choose car simulation also (Yes)

aadi-mishra, 2018-06-07:
Error persists. Tried both methods as you suggested. In my case, UE doesn't prompt me twice. check this error from UE.
[image]

aadi-mishra, 2018-06-07:
@kshitiz38 when the vehicles spawn in the environment, do both of them spawn at the same location?

kshitiz38, 2018-06-07:
@aadi-mishra ..This seem like an issue with your UE4 setup...Are you on lastest commit, The AirSim supports UE 4.18. If you are working with some other version or custom environment please make sure that you are able to build the official release and run default Environment Blocks.

I do not have a windows environment so I am unable to reproduce this error..maybe someone else on the thread can help to see if it works for Windows.

Yes both vehicles are spawned at same location(X,Y) but their Z values are different..This can be easily modified in setupVehiclesAndCamera() by modifying variables car_transform and multirotor_transform

foorenxiang, 2018-06-11:
I've no issues running this on Windows 10 on UE 4.18.3.  Thanks for the great fork @kshitiz38

aadi-mishra, 2018-06-12:
@kshitiz38  Working now! Thanks a lot.

aadi-mishra, 2018-07-04:
@kshitiz38  Any updates on the new updated Airsim repo for heterogeneous vehicles? I have your repo forked.

alessand01, 2018-12-05:
> Hi! I was able to achieve this, you can check my fork. For my scenario I wanted to be able to control the multirotor using APIs to track the car which I would drive around. I had to make [...]

Hi, I'm using your fork to simulate a drone and car. 
I'm also trying to make the drone follows the car. 
Is there a way to attach the viewport to the drone instead of the car? 

Thanks!

alessand01, 2018-12-10:
> @alessand01 Yes with the new fork just mark the vehicle as fpv in settings.json file.

Thank you!
I have download the new fork but I get a segmentation fault error running the default HelloDrone application.. Any idea?

Thanks!

sousheel, 2019-01-05:
@kshitiz38 I have been trying to run your fork, but have come across a number of problems I am unsure of how to fix. I am experiencing the same seg fault as @alessand01. This only occurs after trying to move the car. I am also interested in the same viewport on the drone and the FPV option doesn't work for me. Do you have any pointers on how I can help to fix these problems? I can give more info on the seg fault as well.

fmazzi, 2019-01-09:
@kshitiz38 I'm also experiencing seg fault  even when no clients are connected. The crash occurs after a very variable period of time. (from 5 sec to some minutes). I have attached some log files about it.

[Diagnostics.txt](https://github.com/Microsoft/AirSim/files/2740180/Diagnostics.txt)
[Blocks.log](https://github.com/Microsoft/AirSim/files/2740183/Blocks.log)

FL2316, 2019-05-17:
I follow your advice, but there are some mistakes that can't be solved. I don't know what to do...seg fault,
Can anybody help me?thanks alot

josepeck, 2019-12-02:
Hi @kshitiz38 , have you tried usign flight controller "PX4" insted of "SimpleFlight" in your fork ??

sheida67, 2020-01-14:
@kshitiz38 
Hi,
When I try to run DQNAgentSim.py I face with this error:
RPCError: rpclib: server could not find function 'setCarControls' with argument count 2.
I use python- 3.6.9 but I don't think so it relates to the python version.
Thanks,

cnun02, 2020-02-27:
my OS : Linux 18.04LTS
Airsim version : 1.2.0
UnrealEngine version : 4.18.3

I've tried the above solutions.
But it did not work properly.

explan brief  possible, I downloaded Airsim from @ kshitiz38's repository and ran ./setup.sh and ./build.sh normally.

After the build was complete, I copied the Plugin folder inside the Unreal folder into UE4 4.18.3 version project folder.

Airsim plugin has been added normally with the same settings as when using Microsoft / Airsim.

In the UE4,  I choosed game mode to Airsim game mode.

and I was set up setting.js file like this.

{
    "SettingsVersion": 1.2,
    "SimMode": "Both",
    "Vehicles": {
        "Car1": {
          "VehicleType": "PhysXCar",
          "X": 0, "Y": 0, "Z": -2
        },
        "Drone1": {
          "VehicleType": "SimpleFlight",
          "X": 0, "Y": 0, "Z": -5,
      "Yaw": 90
        }
  }
}

But it is impossible to execute.

and pop up message like this,
" Simmode is not valid. "Both" "

what should I do? :(
please help me...

wenwenla, 2020-03-04:
@fmazzi Do you have any idea about these errors. When using the both mode on windows, I encountered some runtime errors as yours.
Here is some additional information.
[image]

Akshath-Singhal, 2020-06-20:
@kshitiz38 I tried using your both-1.2.1 branch with Unreal 4.24 on ubuntu 18.04 and got the following error:

`The project could not be compiled. Would you like to open it in Null Source Code Access?

Running /home/akshath/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Development Linux -Project="/home/akshath/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -TargetType=Editor -Progress -NoHotReloadFromIDE
Scale at X=0.000000, Y=0.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))
Scale at X=960.000000, Y=540.000000: 1.000000 (monitor=#0, HDPI=96.000000 (horz scale: 1.000000), VDPI=96.000000 (vert scale: 1.000000))

Running Mono...

Fixing inconsistent case in filenames.
Setting up Mono
~/UnrealEngine/Engine ~/UnrealEngine/Engine/Binaries/Linux
Using 'git status' to determine working set for adaptive non-unity build (/home/akshath/UnrealEngine).
Creating makefile for BlocksEditor (no existing makefile)
WARNING: Library 'stdc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
WARNING: Library 'supc++' was not resolvable to a file when used in Module 'AirSim', assuming it is a filename and will search library paths for it. This is slow and dependency checking will not work for it. Please update reference to be fully qualified alternatively use PublicSystemLibraryPaths if you do intended to use this slow path to suppress this warning.
@progress push 5%
Creating makefile for UnrealHeaderTool (no existing makefile)
@progress push 5%
@progress pop
Waiting for 'git status' command to complete
------- Build details --------
Using toolchain located at '/home/akshath/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu'.
Using clang (/home/akshath/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/clang++) version '8.0.1' (string), 8 (major), 0 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using lld linker
Using llvm-ar : /home/akshath/UnrealEngine/Engine/Extras/ThirdPartyNotUE/SDKs/HostLinux/Linux_x64/v15_clang-8.0.1-centos7/x86_64-unknown-linux-gnu/bin/llvm-ar
Using fast way to relink  circularly dependent libraries (no FixDeps).
------------------------------
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/home/akshath/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/home/akshath/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
/home/akshath/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/CameraDirector.h(24) : LogCompile: Error: 'ECameraDirectorMode' does not have a 0 entry! (This is a problem when the enum is initalized by default)
`

Can you help me with the debugging?

sandyUni, 2020-10-29:
Is there any progress?
To my best of my knowledge, since the implementations of the ground and aerial vehicles in Airsim are totally different, they have different data retrieve policies (for groud vehicle, Airsim fetches the sensor data once at each unreal render frame, while for aerial vehicle, it fetches data every sample time.  So does your solution addresses this problem?

kartikeya13, 2021-05-09:
@sandyUni Hii, did you able to do it? I tried it with everything available but looks like some of the EIGEN policies has updated which hinders the successful vuild.

danieto98, 2021-05-18:
Hello, to anyone still interested in this:

I followed the work of @kshitiz38 from [his repository](https://github.com/kshitiz38/AirSim) and what I found [here](https://github.com/IdoMatan/AirsimAugmentedFiles) and I managed to get a joint simulation to work. I also went ahead and updated the ROS wrapper for AirSim to account for this.

You can find my repository [here](https://github.com/danieto98). It should be compatible with the latest version of AirSim.

kartikeya13, 2021-05-20:
hello daniel,
Thanks for sharing. Is it tested on windows as well?

kartikeya13, 2021-05-20:
> @kartikeya13 <https://github.com/kartikeya13> Unfortunately it has not
> been tested on Windows yet, I will leave that up to you. My latest commit
> makes use of OpenCV and has nothing to do with the joint simulation, you
> can roll back to the previous one to test it without having to worry about
> Windows compilation with OpenCV

Sure, if will do it. If its for Linux I will use your fork in my current
work. thanks a lot

kartikeya13, 2021-07-13:
Hii Daniel, Thanks for the repo.
But, I am not being able to move car after rendering both car and drone
using your repo.
Any help on this would be appreciated
Thanks

danieto98, 2021-07-13:
Hi! Can you give me a bit more information?
How are you trying to move the car? Using a joystick or the APIs? Can I see your config file? If using the APIs, did you enable API control etc?
I am able to drive it with the APIs without any issues and do it from ROS as well. Here's my config file:
```
{
 "SettingsVersion": 1.2,
 "SimMode": "Both",
 "Vehicles": {
    "drone_1": {
      "VehicleType": "SimpleFlight",
      "AllowAPIAlways": true,
      "RC": {
        "RemoteControlID": -1,
        "AllowAPIWhenDisconnected": false
      },
      "X": -0.9, "Y": 0.05, "Z": -1.5,
      "Pitch": 0, "Roll": 0, "Yaw": 0
    },
    "Car1": {
       "VehicleType": "PhysXCar",
       "AllowAPIAlways": true,
       "X": 0, "Y": 0, "Z": 0.5, "Yaw": 0
     }
 }
}
```

You can also try [this repo](https://github.com/ACENTAURI-INRIA/AirSim) I've been working on lately which has some other features, but the joint simulation should work the same way.

kartikeya13, 2021-07-15:
Thanks, It was actually the nvidia driver's problem. I had to update them
to control the actors. But I noticed that I am able to use APIs to perform
some moving operations using a car, but when i use drone API along with
car's, the rpc package is not able to determine the servers. Attached is
the error log.

danieto98, 2021-07-15:
I can't see the attachment (I think e-mail reply does not allow you to send them).
I do have an idea what the problem could be though.

When using the APIs, you should create two different RPC clients, one for each vehicle type.
You then need to use the corresponding client for every vehicle you want to control.

If you're using C++, for instance:
```
auto drone_client = new msr::airlib::MultirotorRpcLibClient("localhost", 41451);
auto car_client = new msr::airlib::CarRpcLibClient("localhost", 41452);

// Your application code here

delete drone_client;
delete car_client;
```

Note that each RPC client uses a different port number, you should use those exact numbers for the drone and car client respectively (or change them in the file Unreal/Plugins/AirSim/Source/Vehicles/Multirotor/SimModeWorldBoth.cpp inside the definition of createApiServer()).

kartikeya13, 2021-07-16:
Got It, Thanks. I am attaching the error screenshot for everyone's reference. 
Also, can you please let me know about how to assign the RPC ports if we are using python APIs?
[image]

zjlanhe, 2021-09-01:
> Hello, to anyone still interested in this:
>
> I followed the work of @kshitiz38 from [his repository](https://github.com/kshitiz38/AirSim) and what I found [...]

dear danieto98,
i have an issue when i use your branch. could you give some suggestion?
I play it in UE with both mode(which is set in settings.json), muti cars and muti drones can normally displayed in UEEditor，it seems good.
but i don't do nothing ,just wait for a moment(sometimes one minute,sometimes several minutes), UEEditor will crash. 
here is the log when crashed.  it's difficult for me to debug it.  i don't have any idea.
[image]

ertugrultaparci, 2022-03-13:
Hello, is there anyone totally solved this issues? Can you explain in detail how to achieve this? Can I do this in Windows also? Do I use both car and multirotor in a different simulation environment (like **Plains**) ? I really need these information to prepare a term project.

Hao-HUST, 2022-10-18:
> Hello, to anyone still interested in this:
>
> I followed the work of @kshitiz38 from [his repository](https://github.com/kshitiz38/AirSim) and what I found [...]

Hi, I tried to use your modified AirSim to simulate the car and drone together, but Unreal Engine 4 always crashed when I ran Blocks.uproject file in the /AirSim/Unreal/Environments/Block folder, the crash reporter is described as follows: 

> LoginId:ea90bb26ff37438e98045d40ec4b1c8e-000003e8
EpicAccountId:
Caught signal
libUE4Editor-AirSim.so!msr::airlib::DelayLine<msr::airlib::GpsBase::Output>::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/common/DelayLine.hpp:54]
libUE4Editor-AirSim.so!msr::airlib::GpsSimple::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/sensors/gps/GpsSimple.hpp:60]
libUE4Editor-AirSim.so!msr::airlib::UpdatableContainer<msr::airlib::SensorBase*>::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/common/UpdatableContainer.hpp:60]
libUE4Editor-AirSim.so!msr::airlib::SensorCollection::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/sensors/SensorCollection.hpp:83]
libUE4Editor-AirSim.so!CarPawnSimApi::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/Vehicles/Car/CarPawnSimApi.cpp:169]
libUE4Editor-AirSim.so!PawnSimApi::pawnTick(float) [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/PawnSimApi.cpp:77]
libUE4Editor-AirSim.so!ACarPawn::Tick(float) [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/Vehicles/Car/CarPawn.cpp:260]
libUE4Editor-Engine.so!AActor::TickActor(float, ELevelTick, FActorTickFunction&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/Actor.cpp:1007]
libUE4Editor-Engine.so!FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/Actor.cpp:171]
libUE4Editor-Engine.so!FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr<FGraphEvent> const&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:283]
libUE4Editor-Engine.so!TGraphTask<FTickFunctionTask>::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:849]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:692]
libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:586]
libUE4Editor-Core.so!FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr<FGraphEvent>, TInlineAllocator<4u, TSizedDefaultAllocator<32> > > const&, ENamedThreads::Type) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:1477]
libUE4Editor-Engine.so!FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:557]
libUE4Editor-Engine.so!FTickTaskManager::RunTickGroup(ETickingGroup, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1522]
libUE4Editor-Engine.so!UWorld::Tick(ELevelTick, float) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1558]
libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1692]
libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:411]
UE4Editor!FEngineLoop::Tick() [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4844]
UE4Editor!GuardedMain(char16_t const*) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
libUE4Editor-UnixCommonStartup.so!CommonUnixMain(int, char**, int (*)(char16_t const*), void (*)()) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Unix/UnixCommonStartup/Private/UnixCommonStartup.cpp:264]
libc.so.6!__libc_start_main(+0xe6)
UE4Editor!_start()

Do you have any suggestions about that? Is that possible caused by the mismatch between AirSim and Unreal Version? (My current Unreal Version is  4.25.0)

coolkesword, 2023-06-02:
> > Hello, to anyone still interested in this:
> > I followed the work of @kshitiz38 from [his repository](https://github.com/kshitiz38/AirSim) and what I found [here](https://github.com/IdoMatan/AirsimAugmentedFiles) and I managed to get a joint simulation to work. I also went ahead and updated the ROS wrapper for AirSim to account for this.
> > You can find my repository [here](https://github.com/danieto98). It should be compatible with the latest version of AirSim.
> 
> Hi, I tried to use your modified AirSim to simulate the car and drone together, but Unreal Engine 4 always crashed when I ran Blocks.uproject file in the /AirSim/Unreal/Environments/Block folder, the crash reporter is described as follows:
> 
> > LoginId:ea90bb26ff37438e98045d40ec4b1c8e-000003e8
> > EpicAccountId:
> > Caught signal
> > libUE4Editor-AirSim.so!msr::airlib::DelayLinemsr::airlib::GpsBase::Output::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/common/DelayLine.hpp:54]
> > libUE4Editor-AirSim.so!msr::airlib::GpsSimple::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/sensors/gps/GpsSimple.hpp:60]
> > libUE4Editor-AirSim.so!msr::airlib::UpdatableContainermsr::airlib::SensorBase*::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/common/UpdatableContainer.hpp:60]
> > libUE4Editor-AirSim.so!msr::airlib::SensorCollection::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/AirLib/include/sensors/SensorCollection.hpp:83]
> > libUE4Editor-AirSim.so!CarPawnSimApi::update() [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/Vehicles/Car/CarPawnSimApi.cpp:169]
> > libUE4Editor-AirSim.so!PawnSimApi::pawnTick(float) [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/PawnSimApi.cpp:77]
> > libUE4Editor-AirSim.so!ACarPawn::Tick(float) [/home/hao/AirSim-master/Unreal/Environments/Blocks 4.25 - 2/Plugins/AirSim/Source/Vehicles/Car/CarPawn.cpp:260]
> > libUE4Editor-Engine.so!AActor::TickActor(float, ELevelTick, FActorTickFunction&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/Actor.cpp:1007]
> > libUE4Editor-Engine.so!FActorTickFunction::ExecuteTick(float, ELevelTick, ENamedThreads::Type, TRefCountPtr const&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/Actor.cpp:171]
> > libUE4Editor-Engine.so!FTickFunctionTask::DoTask(ENamedThreads::Type, TRefCountPtr const&) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:283]
> > libUE4Editor-Engine.so!TGraphTask::ExecuteTask(TArray<FBaseGraphTask*, TSizedDefaultAllocator<32> >&, ENamedThreads::Type) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Public/Async/TaskGraphInterfaces.h:849]
> > libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksNamedThread(int, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:692]
> > libUE4Editor-Core.so!FNamedTaskThread::ProcessTasksUntilQuit(int) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:586]
> > libUE4Editor-Core.so!FTaskGraphImplementation::WaitUntilTasksComplete(TArray<TRefCountPtr, TInlineAllocator<4u, TSizedDefaultAllocator<32> > > const&, ENamedThreads::Type) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Core/Private/Async/TaskGraph.cpp:1477]
> > libUE4Editor-Engine.so!FTickTaskSequencer::ReleaseTickGroup(ETickingGroup, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:557]
> > libUE4Editor-Engine.so!FTickTaskManager::RunTickGroup(ETickingGroup, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/TickTaskManager.cpp:1522]
> > libUE4Editor-Engine.so!UWorld::Tick(ELevelTick, float) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Engine/Private/LevelTick.cpp:1558]
> > libUE4Editor-UnrealEd.so!UEditorEngine::Tick(float, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Editor/UnrealEd/Private/EditorEngine.cpp:1692]
> > libUE4Editor-UnrealEd.so!UUnrealEdEngine::Tick(float, bool) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Editor/UnrealEd/Private/UnrealEdEngine.cpp:411]
> > UE4Editor!FEngineLoop::Tick() [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Launch/Private/LaunchEngineLoop.cpp:4844]
> > UE4Editor!GuardedMain(char16_t const*) [/media/hao/004d31d5-15fd-4652-9c9e-0304bcefff7c/project/drone/UnrealEngine-4.25.0-release/Engine/Source/Runtime/Launch/Private/Launch.cpp:171]
[... cut ...]
    libUE4Editor-Engine.so!AActor::TickActor(float, ELevelTick, FActorTickFunction&) [/home/dasa/src/UnrealEngine-4.27-plus/Engine/Source/Runtime/Engine/Private/Actor.cpp:1090]
```
 
I followed the instruction shown in: [link](https://idoglanz.medium.com/setting-up-microsoft-airsim-to-simulate-a-drone-and-car-together-708079b2d0f). My computer setup is Ubuntu 18.04, with the latest Airsim version and Unreal Engine 4.27.2.

Really appreciate that if you could give any suggestion regarding this issue.

Thanks
Jie

ssaigarimella, 2025-02-10:
@coolkesword @Hao-HUST by any chance were you able to solve this? I am getting the same segmentation fault from DelayLine.hpp