microsoft/AirSim #509: "Eigen/Dense":No such file or directory [c:\CHUANGYE\AirSim\AirLib\AirLib.vcxproj]

fenfeibani, 2017-10-03:
when i run build.cmd in VS2015  x64 Native Tools Command Prompt, it sent me the error message:
"Eigen/Dense":No such file or directory [c:\CHUANGYE\AirSim\AirLib\AirLib.vcxproj]

 My system is  win7+vs2015!

sytelus, 2017-10-04:
I'm guessing its because of old or no powershell on Windows 7. Could you please install WMM 4.0 and try again? https://www.microsoft.com/en-us/download/details.aspx?id=40855

jldavis2, 2017-10-06:
Having the same issue as well. Downloaded and updated the WMM per the link and verified it is up to date (brought up a powershell prompt, typed in "$PSVersionTable.PSVersion" and it's showing major version 4), still running into the same issue though..

sytelus, 2017-10-07:
I don't have Win7 installation so you will need to debug this... The error is likely happening because below command isn't working for some reason. You can open VS2015 x64 command prompt, navigate to AirSim folder and then type below commands and see what happens...

```
    powershell -command "& { iwr http://bitbucket.org/eigen/eigen/get/3.3.2.zip -OutFile eigen3.zip }"
    powershell -command "& { Expand-Archive -Path eigen3.zip -DestinationPath AirLib\deps }"
```

jldavis2, 2017-10-09:
Hi,

The first command ran fine, however I got the following error when running
the second command:

C:\Users\davisjl\dev\Repositories\AirSim>powershell -command "& {
Expand-Archive -Path eigen3.zip -DestinationPath AirLib\deps }"
Expand-Archive : The term 'Expand-Archive' is not recognized as the name of
a cmdlet, function, script file, or operable program. Check the spelling of
the name, or if a path was included, verify that the path is correct and
try again.

At line:1 char:5
+ & { Expand-Archive -Path eigen3.zip -DestinationPath AirLib\deps }
+     ~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (Expand-Archive:String) [],
Comm
   andNotFoundException
    + FullyQualifiedErrorId : CommandNotFoundException

In a quick google search, it doesn't look like the command Expand-Archive
is available for WMM 4, at least according to the MS source I was looking
at. I'm going to try installing 5.1 (according to the same source that
version has Expand-Archive), I'll let you know how it goes!

sytelus, 2017-10-10:
Thanks. Yes, 5.1 is default in Win10 and hopefully that should work.

jldavis2, 2017-10-10:
Yup upgrading to 5.1 managed to fix the issue.

One quick question, I managed to run build.cmd without any apparent issues,
the unreal/plugins/airsim looked to be built and I copied the folder into
the engine/plugins folder. However when I start up Unreal Engine I am
getting an error saying "plugin failed to load because module AirSim could
not be found". Inside the Airsim folder I see an AirLib folder with a
AirLib.lib, should there be an AirSim.dll somewhere in there? (ie is the
build failing before it reaches the end?)

Thanks again for the help.

Jim

fenfeibani, 2017-10-25:
Thank you for your help, i have solved this problem!

closed 2017-11-14 by sytelus