microsoft/AirSim #512: Downtown env for Multirotor

Kjell-K, 2017-10-04:
@sytelus It would be great to have the Downtown binaries env fully compatible with the MultirotorClient() as well. 
Also setting up Collisions with buildings as static meshes would be necessary then.

Downtown is a much more attractive environment for Copter than Neighborhood.

sytelus, 2017-10-06:
This is working fine on my m/c. What do you see?

Kjell-K, 2017-10-06:
I was just missing the information text on the upper left, but it seems it is removed for all envs?

Therefore the main point is: 

> **Also setting up Collisions with buildings as static meshes would be necessary then.**

sytelus, 2017-10-06:
Current binaries are in shipping config. I think I should revert them back to Develop config so you can see log messages. I will check out the collisons.

sytelus, 2017-11-14:
We have newer City environment now. Also all log text should be working in new release.

closed 2017-11-14 by sytelus