microsoft/AirSim #514: Bug report: car unable to go reverse in v1.1.2.x

kaihuchen, 2017-10-05:
Run hello_car.py under the latest V1.1.2.x Windows binaries in the AirSimNeighbourhood environment, and find that the car goes forward just fine, but does not go in reverse at all.

sytelus, 2017-10-06:
Using keyboard or API? Just tested using down arrow key as well as API and it seems to be working fine. Note that after you call enableApiControl(true), manual control is disabled until you call  enableApiControl(false).

kaihuchen, 2017-10-06:
This is using the API. As mentioned above, the problem can be reproduced by simply running the official hello_car.py as-is without change.

sytelus, 2017-10-07:
Thanks for reporting. This is now fixed in latest source and binaries.

closed 2017-10-07 by sytelus