microsoft/AirSim #517: Unreal 4.16 failed to open blocks environment in unreal

raymondchua, 2017-10-06:
I am trying to open Blocks.uproject in Unreal Engine 4.16.3 and I got a prompt that 
libUE4Editor-Block.so
libUE4Editor-AirSim.so
are missing or built with different version. When I try to rebuild them, I get the following error:

```
@progress push 5%
@progress pop
Performing 3 actions (8 in parallel)
[1/3] Compile Module.AirSim.cpp
In file included from /work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:30:
/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:125:10: fatal error: 
      'libudev.h' file not found
#include <libudev.h>
         ^
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 7.92 seconds (Local executor: 0.00 seconds)
LogInit:Warning: Still incompatible or missing module: libUE4Editor-Blocks.so
LogInit:Warning: Still incompatible or missing module: libUE4Editor-AirSim.so
LogExit: Preparing to exit.
LogModuleManager: Shutting down and abandoning module DesktopPlatform (34)
LogModuleManager: Shutting down and abandoning module AnimationModifiers (32)
LogModuleManager: Shutting down and abandoning module PropertyEditor (31)
LogModuleManager: Shutting down and abandoning module AudioEditor (28)
LogModuleManager: Shutting down and abandoning module TextureCompressor (26)
LogModuleManager: Shutting down and abandoning module ShaderCore (24)
LogModuleManager: Shutting down and abandoning module Landscape (22)
LogModuleManager: Shutting down and abandoning module SlateRHIRenderer (20)
LogModuleManager: Shutting down and abandoning module OpenGLDrv (18)
LogModuleManager: Shutting down and abandoning module AnimGraphRuntime (16)
LogModuleManager: Shutting down and abandoning module Renderer (14)
LogModuleManager: Shutting down and abandoning module Engine (12)
LogModuleManager: Shutting down and abandoning module CoreUObject (10)
LogModuleManager: Shutting down and abandoning module NetworkFile (8)
LogModuleManager: Shutting down and abandoning module StreamingFile (6)
LogModuleManager: Shutting down and abandoning module SandboxFile (4)
LogModuleManager: Shutting down and abandoning module PakFile (2)
LogExit: Exiting.
LogInit: Tearing down SDL.
Exiting abnormally (error code: 1)
```
Any thoughts on how I can fix it?

sytelus, 2017-10-07:
Are you using Ubuntu 16.04? Did you ran setup.sh first?

raymondchua, 2017-10-09:
@sytelus, I have rerun setup.sh and now when I tried to compile the blocks environment, I get an error that says that the project cannot be compiled.

```
Running /work/raymond/idp/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE
Setting swap interval to 'Immediate'

Running Mono...

/work/raymond/idp/UnrealEngine/Engine /work/raymond/idp/UnrealEngine/Engine/Binaries/Linux
Performing full C++ include scan (building a new target)
Creating makefile for Blocks (no existing makefile)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
@progress push 5%
Building UnrealHeaderTool...
Creating makefile for UnrealHeaderTool (no existing makefile)
Using clang version '3.9.1' (string), 3 (major), 9 (minor), 1 (patch)
Using bundled libc++ standard C++ library.
Using fast way to relink  circularly dependent libraries (no FixDeps).
Target is up to date
Total build time: 3.15 seconds (NoActionsToExecute executor: 0.00 seconds)
Parsing headers for BlocksEditor
  Running UnrealHeaderTool "/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Blocks.uproject" "/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Intermediate/Build/Linux/B4D820EA/BlocksEditor/Development/BlocksEditor.uhtmanifest" -LogCmds="loginit warning, logexit warning, logdatabase error" -Unattended -WarningsAsErrors
LogPlatformFile: Not using cached read wrapper
LogInit: Display: RandInit(1701502845) SRandInit(1701502845).
LogTaskGraph: Started task graph with 4 named threads and 7 total threads with 1 sets of task threads.
This project requires the 'UdpMessaging' plugin, which is not compatible with the current engine version.
Would you like to disable it? You will no longer be able to open any assets created using it.
LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin SubversionSourceControl
[... 1293 lines cut ...]
[2017.10.09-11.21.49:885][  0]LogCompile: Preparsed module /Script/Localization containing 3 files(s) in 0.00 secs.
[2017.10.09-11.21.49:890][  0]LogCompile: Preparsed module /Script/AutomationController containing 2 files(s) in 0.00 secs.
[2017.10.09-11.21.49:891][  0]LogCompile: Preparsed module /Script/MaterialBaking containing 1 files(s) in 0.00 secs.
[2017.10.09-11.21.49:895][  0]LogCompile: Preparsed module /Script/AITestSuite containing 12 files(s) in 0.00 secs.
[2017.10.09-11.21.49:896][  0]LogCompile: Preparsed module /Script/AutomationWindow containing 1 files(s) in 0.00 secs.
[2017.10.09-11.21.49:897][  0]LogCompile: Preparsed module /Script/HTML5PlatformEditor containing 2 files(s) in 0.00 secs.
[2017.10.09-11.21.49:898][  0]LogCompile: Preparsed module /Script/AndroidPlatformEditor containing 1 files(s) in 0.00 secs.
[2017.10.09-11.21.49:901][  0]LogCompile: Preparsed module /Script/LogVisualizer containing 6 files(s) in 0.00 secs.
[2017.10.09-11.21.49:901][  0]LogCompile: Preparsed module /Script/LinuxTargetPlatform containing 1 files(s) in 0.00 secs.
[2017.10.09-11.21.49:902][  0]LogCompile: Preparsed module /Script/TextureEditor containing 1 files(s) in 0.00 secs.
[... cut ...]
[2017.10.09-11.21.49:984][  0]LogCompile: Preparsed module /Script/Sequencer containing 5 files(s) in 0.01 secs.
[2017.10.09-11.21.49:984][  0]LogCompile: Preparsed module /Script/HierarchicalLODOutliner containing 2 files(s) in 0.00 secs.
[2017.10.09-11.21.49:985][  0]LogCompile: Preparsed module /Script/PixelInspectorModule containing 1 files(s) in 0.00 secs.
[2017.10.09-11.21.49:988][  0]LogCompile: Preparsed module /Script/SequenceRecorder containing 6 files(s) in 0.00 secs.
[2017.10.09-11.21.49:991][  0]LogCompile: Preparsed module /Script/MovieSceneTools containing 5 files(s) in 0.00 secs.

```

raymondchua, 2017-10-10:
Hi, I have switch to Unreal 4.17 and when I now try to open the Block environment, I got the following error that 'Blocks could not be compiled.Try rebuilding from source manually.'

```
/work/raymond/idp/UnrealEngine$ LogPluginManager: Mounting plugin AssetManagerEditor
LogPluginManager: Mounting plugin PluginBrowser
LogPluginManager: Mounting plugin FacialAnimation
LogPluginManager: Mounting plugin GameplayTagsEditor
LogPluginManager: Mounting plugin AudioCapture
LogPluginManager: Mounting plugin MobileLauncherProfileWizard
LogPluginManager: Mounting plugin SpeedTreeImporter
LogPluginManager: Mounting plugin OnlineSubsystemIOS
LogPluginManager: Mounting plugin OnlineSubsystemNull
LogPluginManager: Mounting plugin OnlineSubsystemGooglePlay
LogPluginManager: Mounting plugin OnlineSubsystemUtils
LogPluginManager: Mounting plugin OnlineSubsystem
LogPluginManager: Mounting plugin HTML5Networking
LogPluginManager: Mounting plugin AlembicImporter
LogPluginManager: Mounting plugin CharacterAI
LogPluginManager: Mounting plugin SteamVR
LogPluginManager: Mounting plugin WindowsMoviePlayer
LogPluginManager: Mounting plugin ProceduralMeshComponent
LogPluginManager: Mounting plugin AppleMoviePlayer
LogPluginManager: Mounting plugin IOSDeviceProfileSelector
LogPluginManager: Mounting plugin ImmediatePhysics
LogPluginManager: Mounting plugin AndroidMoviePlayer
LogPluginManager: Mounting plugin OculusVR
LogPluginManager: Mounting plugin GoogleCloudMessaging
LogPluginManager: Mounting plugin CustomMeshComponent
LogPluginManager: Mounting plugin LocationServicesBPLibrary
LogPluginManager: Mounting plugin AndroidPermission
LogPluginManager: Mounting plugin ExampleDeviceProfileSelector
LogPluginManager: Mounting plugin RuntimePhysXCooking
LogPluginManager: Mounting plugin ArchVisCharacter
LogPluginManager: Mounting plugin LinuxDeviceProfileSelector
LogPluginManager: Mounting plugin MobilePatchingUtils
LogPluginManager: Mounting plugin CableComponent
LogPluginManager: Mounting plugin PhysXVehicles
LogPluginManager: Mounting plugin AndroidDeviceProfileSelector
LogPluginManager: Mounting plugin NullSourceCodeAccess
LogPluginManager: Mounting plugin KDevelopSourceCodeAccess
LogPluginManager: Mounting plugin GitSourceControl
LogPluginManager: Mounting plugin CodeLiteSourceCodeAccess
LogPluginManager: Mounting plugin UObjectPlugin
LogPluginManager: Mounting plugin VisualStudioSourceCodeAccess
LogPluginManager: Mounting plugin PerforceSourceControl
LogPluginManager: Mounting plugin SubversionSourceControl
LogPluginManager: Mounting plugin XCodeSourceCodeAccess
LogPluginManager: Mounting plugin AvfMedia
LogPluginManager: Mounting plugin AndroidMedia
LogPluginManager: Mounting plugin MediaPlayerEditor
LogPluginManager: Mounting plugin WmfMedia
LogPluginManager: Mounting plugin LightPropagationVolume
LogPluginManager: Mounting plugin Paper2D
LogPluginManager: Mounting plugin TcpMessaging
LogPluginManager: Mounting plugin UdpMessaging
LogPluginManager: Mounting plugin MatineeToLevelSequence
LogPluginManager: Mounting plugin LevelSequenceEditor
LogPluginManager: Mounting plugin AirSim
LogInit: Warning: Incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Incompatible or missing module: libUE4Editor-AirSim.so
STUBBED: FDesktopPlatformLinux::GetNativeFeedbackContext at /work/raymond/idp/UnrealEngine/Engine/Source/Developer/DesktopPlatform/Private/Linux/DesktopPlatformLinux.cpp:269 (GetNativeFeedbackContext)
Running /work/raymond/idp/UnrealEngine/Engine/Binaries/DotNET/UnrealBuildTool.exe Blocks Development Linux -project="/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Blocks.uproject" -editorrecompile -progress -NoHotReloadFromIDE

Running Mono...

/work/raymond/idp/UnrealEngine/Engine /work/raymond/idp/UnrealEngine
@progress push 5%
@progress pop
Performing 3 actions (8 in parallel)
[1/3] Compile Module.AirSim.cpp
In file included from /work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Intermediate/Build/Linux/B4D820EA/UE4Editor/Development/AirSim/Module.AirSim.cpp:31:
/work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Source/SimJoyStick/SimJoyStick.cpp:125:10: fatal error: 
      'libudev.h' file not found
#include <libudev.h>
         ^
1 error generated.
ERROR: UBT ERROR: Failed to produce item: /work/raymond/idp/AirSim/Unreal/Environments/Blocks/Plugins/AirSim/Binaries/Linux/libUE4Editor-AirSim.so
Total build time: 8.00 seconds (Local executor: 0.00 seconds)
LogInit: Warning: Still incompatible or missing module: libUE4Editor-Blocks.so
LogInit: Warning: Still incompatible or missing module: libUE4Editor-AirSim.so

```

raymondchua, 2017-10-13:
Fix by reinstalling based on the new AirSim release.

closed 2017-10-13 by raymondchua

SaiVineethJ, 2018-01-19:
Even i am having the same problem can u tell, do i need to reinstall the unreal engine again and if yes where can i find it based on the new AirSim release?