microsoft/AirSim #518: Linux build error: linker command failed

ChristophTobler, 2017-10-06:
I use Ubuntu 16.04 and `./setup.sh` finished successfully. When I run `./build.sh` I get
```
clang: error: linker command failed with exit code 1 (use -v to see invocation)
HelloDrone/CMakeFiles/HelloDrone.dir/build.make:97: recipe for target 'output/bin/HelloDrone' failed
make[2]: *** [output/bin/HelloDrone] Error 1
CMakeFiles/Makefile2:385: recipe for target 'HelloDrone/CMakeFiles/HelloDrone.dir/all' failed
make[1]: *** [HelloDrone/CMakeFiles/HelloDrone.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

mhct, 2017-10-07:
Hi, 

The same happened to me. I solved it by checking out an older commit from Airsim.
`git checkout 1fe5e40`
`build.sh`

and it works

ChristophTobler, 2017-10-07:
Yeah, I switched to an older commit as well. Thanks though!

sytelus, 2017-10-07:
Thanks for reporting. This is now fixed in latest version.

closed 2017-10-07 by sytelus