microsoft/AirSim #519: Unable to Generate Blocks Worlds Example Environment

JeremyBYU, 2017-10-06:
I attempted to follow instructions as outlined [here](https://github.com/Microsoft/AirSim/blob/master/docs/unreal_blocks.md). It looks like copying the built AirSim plugin was succesful (those results are not shown), it just failed during the generation of the solution file. It seems to be looking for something in the Registry when running `GenerateProjectFiles.bat`. Thanks for any help!
```
 cmd /c GenerateProjectFiles.bat

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> setlocal

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> del /q gen_temp.txt
Could Not Find C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks\gen_temp.txt

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> powershell -command "& { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' -Name 'Icon' ).'Icon' }"  1>gen_temp.txt

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> set /p gen_bin= 0<gen_temp.txt

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> del /q gen_temp.txt
[Get-ItemProperty] was unexpected at this time.

@ C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
>               Get-ItemProperty : Cannot find path 'HKEY_CLASSES_ROOT\Unreal.ProjectFile\shell\rungenproj' because it does not exist.
At line:1 char:6
+ & { (Get-ItemProperty 'Registry::HKEY_CLASSES_ROOT\Unreal.ProjectFile ...
+      ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : ObjectNotFound: (HKEY_CLASSES_RO...hell\rungenproj:String) [Get-ItemProperty], ItemNotFo
   undException
    + FullyQualifiedErrorId : PathNotFound,Microsoft.PowerShell.Commands.GetItemPropertyCommand
  /projectfiles C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks\%f

Jerem@LAPTOP-RIC64UK2 C:\Users\Jerem\Documents\UMICH\Research\AirSim\Unreal\Environments\Blocks
> pause
Press any key to continue . . .

```

sytelus, 2017-10-06:
This is strange, I have never seen this or reported. Could you please confirm if you are running this from Visual Studio x64 Command Prompt? Another possibility is that Unreal Engine is not installed properly on your m/c. On my m/c I have below registry keys after install but I suppose your m/c doesn't:

[image]

So one thing to try is simply reinstall UE4.

Another thing to try is right click on .uproject file and you should see "Generate Project Files" option. Try clicking on it. If you don't then its definitely install issue.

JeremyBYU, 2017-10-06:
Yep, you were absolutely right. These registry entries did not exist for me (`Unreal.ProjectFile`). I did a reboot, launched Epic Launcher again and a prompt appeared saying "Unreal Engine files are not associated, would you like to associate them?".  I said yes, and now the keys have been magically created!

Thanks for you quick response and help!

closed 2017-10-06 by JeremyBYU

mayank-technical, 2019-04-16:
Hi simply run UnrealVersionSelector-Win64-Shipping.exe located at ...\Engine\Binaries\Win64\ and voila check regedit you will see all the entries as sometimes you might also need to build UE from source

harnix, 2020-03-14:
I encountered a similar project and creating a default .uproject helped me resolve this issue.

leommiranda, 2020-04-05:
I found a similar challenge. The answer from @leighswift in this [forum ](https://answers.unrealengine.com/questions/579855/this-file-does-not-have-a-program-associated.html?sort=votes#) solved my problem. By now, it's the most upvoted answer.