microsoft/AirSim #520: HelloDrone causes build.sh to fail in AirSim

raymondchua, 2017-10-06:
Hi, 
I was executing the command './build.sh' for AirSim and I got the following error:

```
./build.sh
+ git submodule update --init --recursive
+ [[ -d ./llvm-build/output/lib ]]
+ export CC=clang-3.9
+ CC=clang-3.9
+ export CXX=clang++-3.9
+ CXX=clang++-3.9
+ [[ -d ./AirLib/deps/eigen3/Eigen ]]
+ build_dir=build_debug
+ echo 'putting build in build_debug folder, to clean, just delete the directory...'
putting build in build_debug folder, to clean, just delete the directory...
+ [[ -f ./cmake/CMakeCache.txt ]]
+ [[ -d ./cmake/CMakeFiles ]]
+ [[ ! -d build_debug ]]
+ pushd build_debug
+ make
[ 21%] Built target AirSim-rpclib
[ 62%] Built target MavLinkCom
[ 77%] Built target AirLib
[ 83%] Built target MavLinkTest
[ 86%] Built target AirLibUnitTests
[ 88%] Linking CXX executable ../output/bin/HelloDrone
clang: warning: argument unused during compilation: '-nostdinc++' [-Wunused-command-line-argument]
CMakeFiles/HelloDrone.dir/work/raymond/idp/AirSim/HelloDrone/main.cpp.o: In function `main':
/work/raymond/idp/AirSim/HelloDrone/main.cpp:30: undefined reference to `msr::airlib::RpcLibClientBase::confirmConnection()'
/work/raymond/idp/AirSim/HelloDrone/main.cpp:34: undefined reference to `msr::airlib::RpcLibClientBase::simGetImages(std::__1::vector<msr::airlib::VehicleCameraBase::ImageRequest, std::__1::allocator<msr::airlib::VehicleCameraBase::ImageRequest> >)'
/work/raymond/idp/AirSim/HelloDrone/main.cpp:62: undefined reference to `msr::airlib::RpcLibClientBase::enableApiControl(bool)'
/work/raymond/idp/AirSim/HelloDrone/main.cpp:76: undefined reference to `msr::airlib::RpcLibClientBase::enableApiControl(bool)'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `MultirotorRpcLibClient':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:33: undefined reference to `msr::airlib::RpcLibClientBase::RpcLibClientBase(std::__1::basic_string<char, std::__1::char_traits<char>, std::__1::allocator<char> > const&, unsigned short, unsigned int)'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `~MultirotorRpcLibClient':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:38: undefined reference to `msr::airlib::RpcLibClientBase::~RpcLibClientBase()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `msr::airlib::MultirotorRpcLibClient::armDisarm(bool)':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:42: undefined reference to `msr::airlib::RpcLibClientBase::getClient()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `msr::airlib::MultirotorRpcLibClient::setSimulationMode(bool)':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:46: undefined reference to `msr::airlib::RpcLibClientBase::getClient()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `msr::airlib::MultirotorRpcLibClient::takeoff(float)':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:50: undefined reference to `msr::airlib::RpcLibClientBase::getClient()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `msr::airlib::MultirotorRpcLibClient::land(float)':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:54: undefined reference to `msr::airlib::RpcLibClientBase::getClient()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o): In function `msr::airlib::MultirotorRpcLibClient::goHome()':
/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:58: undefined reference to `msr::airlib::RpcLibClientBase::getClient()'
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o):/work/raymond/idp/AirSim/AirLib/src/vehicles/multirotor/api/MultirotorRpcLibClient.cpp:62: more undefined references to `msr::airlib::RpcLibClientBase::getClient()' follow
../output/lib/libAirLib.a(MultirotorRpcLibClient.cpp.o):(.data.rel.ro+0x30): undefined reference to `typeinfo for msr::airlib::RpcLibClientBase'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
HelloDrone/CMakeFiles/HelloDrone.dir/build.make:97: recipe for target 'output/bin/HelloDrone' failed
make[2]: *** [output/bin/HelloDrone] Error 1
CMakeFiles/Makefile2:385: recipe for target 'HelloDrone/CMakeFiles/HelloDrone.dir/all' failed
make[1]: *** [HelloDrone/CMakeFiles/HelloDrone.dir/all] Error 2
Makefile:149: recipe for target 'all' failed
make: *** [all] Error 2
```

sytelus, 2017-10-07:
Thanks for reporting. This is now fixed in latest version.

closed 2017-10-07 by sytelus

raymondchua, 2017-10-07:
Thanks. The problem has been fixed.