microsoft/AirSim #523: DroneShell rpc::timeout

exploke, 2017-10-09:
I am attempting to get images of an area in a 360 view using DroneShell. The command I am using is _**==||=> circlebypath -radius 15 -drivetrain 1 -yaw 90 -velocity 1 -z -5**_ but I keep getting the error **_==||=> Error occurred: rpc::timeout: Timeout of 60000ms while calling RPC function 'moveOnPath'_** Is there anyway to fix this timeout issue?

jonyMarino, 2020-08-19:
Closed due to age and incompleteness.
Please feel free to open a new issue if you still have this problem, and include the following (as requested when opening a bug issue)

*Make sure to write all reproduction steps
*Include full error message in text form
*Search issues for error message before filing issue
*Attach screenshot if applicable
*Include code to run if applicable

Thanks!

closed 2020-08-19 by jonyMarino

lknnnli, 2020-12-01:
> I am attempting to get images of an area in a 360 view using DroneShell. The command I am using is _**==||=> circlebypath -radius 15 -drivetrain 1 -yaw 90 -velocity 1 -z -5**_ but I eventually keep getting the error **_==||=> Error occurred: rpc::timeout: Timeout of 60000ms while calling RPC function 'moveOnPath'_** after the drone starts circling around the area. Is there anyway to fix this timeout issue?

hi, have you solved this problem?

sandilyasg, 2024-01-22:
@exploke @lknnnli were you able to solve this problem?