microsoft/AirSim #524: AirSimNeighbourhood not working with RC

robotvin, 2017-10-10:
hello
I have downloaded AirSimNeighbourhood build files. I am using pixhawk with Spectrum DX7S RC. The quadcopter is not responding to RC commands. Anything i m missing. Please guide,

sytelus, 2017-10-10:
Are you using hardware in loop? Is RC transmitter connected to Pixhawk. To use RC with Pixhawk, please see related document on Pixhawk/Arducopter websites such as this: http://ardupilot.org/copter/docs/common-pixhawk-and-px4-compatible-rc-transmitter-and-receiver-systems.html

I would also suggest to try out QGroundControl first to try out Pixhawk configuration before you try out AirSim.

closed 2017-10-10 by sytelus

robotvin, 2017-10-11:
Thanks for reply
Yes the RC is connected with pixhawk and its working well with my own created and build environments using Airsim. But its not working with AirSimNeighbourhood builds.

mchen0037, 2018-01-15:
Hi @robotvin, did you ever find a solution? I'm using a DX8 and having a similar issue.