microsoft/AirSim #525: settings.json file dose not contain much info

robotvin, 2017-10-10:
I want to modify the settings.json file to work with AirSimNeighbourhood environment. When I opened It contains following line  
{
  "see_docs_at": "https://github.com/Microsoft/AirSim/blob/master/docs/settings.md"
}
Please guide me how to change the file.

sytelus, 2017-10-10:
You are likely using old version of AirSim. In version, you should also see "SettingsVersion" key.

Anyway, as the file says look this doc to know more about settings.md: https://github.com/Microsoft/AirSim/blob/master/docs/settings.md

You will need familiarity to edit json format. You can select what you want to change and put those json elements in settings.json.

closed 2017-10-10 by sytelus

robotvin, 2017-10-11:
Ok Thanx for reply
I have one more query! How this setting file created and when?