microsoft/AirSim #526: AirSimLib not compiling together with ZeroMQ

JonathanSchmalhofer, 2017-10-10:
I tried including a **ZeroMQ** Publisher from [libzmq](https://github.com/zeromq/libzmq/) into the **StereoImageGenerator**-Example. For this, I simply added the following include after building the lib itself and adding the dependencies (i.e. include-directories and static libs):

#include "zmq.h"
#include "StereoImageGenerator.hpp"

When adding the ZeroMQ-Header first, I get the following warnings and errors (and many similar following of course):

`D:\AirSim\AirLib\include\common/common_utils/Utils.hpp(144): warning C4003: not enough actual parameters for macro 'min
' [D:\RecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]
D:\AirSim\AirLib\include\common/common_utils/Utils.hpp(144): error C2589: '(': illegal token on right side of '::' [D:\
RecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]
D:\AirSim\AirLib\include\common/common_utils/Utils.hpp(144): error C2059: syntax error: '::' [D:\RecursiveStereoUAV\air
sim_ros_interface\build\airsim_ros_interface.vcxproj]`

When I change the include order as following:

#include "StereoImageGenerator.hpp"
#include "zmq.h"

I do get different warnings and error messages:

`C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ws2def.h(100): warning C4005: 'AF_IPX': macro redefinition [D:\R
ecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock.h(452): note: see previous definition of 'AF_IPX'
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ws2def.h(140): warning C4005: 'AF_MAX': macro redefinition [D:\R
ecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock.h(471): note: see previous definition of 'AF_MAX'
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ws2def.h(177): warning C4005: 'SO_DONTLINGER': macro redefinitio
n [D:\RecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]
  C:\Program Files (x86)\Windows Kits\8.1\Include\um\winsock.h(394): note: see previous definition of 'SO_DONTLINGER'
C:\Program Files (x86)\Windows Kits\8.1\Include\shared\ws2def.h(221): error C2011: 'sockaddr': 'struct' type redefiniti
on [D:\RecursiveStereoUAV\airsim_ros_interface\build\airsim_ros_interface.vcxproj]`

It appears to me, that including `winsock2.h` in ZeroMQ and probably the socket-definitions included/used in (maybe) the rpclib are not compatible?
Any ideas? Did anyone allready encounter this problem?

sytelus, 2017-10-10:
I'm not sure why you are trying ZeroMQ. AirSim does not use ZeroMQ protocol and its API does not work with ZeroMQ. AirSim uses rpclib which in turn uses [msgpack-rpc](https://github.com/msgpack-rpc/msgpack-rpc).

closed 2017-10-10 by sytelus

JonathanSchmalhofer, 2017-10-10:
I tried using ZeroMQ because the receiving application I want to send data to supports ZeroMQ. I did not expect those two libraries clashing that hard. Of course I can switch to rpclib, which I guess will be the easiest for now...