microsoft/AirSim #527: How to define JSON array in CaptureSettings?

Mayankm96, 2017-10-10:
I checked on how to define a JSON array online, and have been trying to do the same in the `settings.json` file. However, the simulator is crashing every time I run it with the new settings. It might be that I am defining the settings the wrong way, but it would help if someone could elaborate on it for me. 

Here is the CaptureSettings I have tried with:
```
"CaptureSettings": [
    {
      "ImageType": [0, 1, 2, 3, 4, 6],
      "Width": 1280,
      "Height": 720,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }
  ],
```

Thank you!

sytelus, 2017-10-13:
You need one item per image type in json array:

```
"CaptureSettings": [
    {
      "ImageType": 1,
      "Width": 1280,
      "Height": 720,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },

    {
      "ImageType": 2,
      "Width": 1280,
      "Height": 720,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },

    {
      "ImageType": 0,
      "Width": 1280,
      "Height": 720,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }

  ],
```

On side note, you are trying to capture at high resolution so expect lower FPS. Also you can (and probably should) omit any settings that you want default values such as FOV.

Mayankm96, 2017-10-14:
Thank you for your reply!  It works now. I'll keep your advice in mind as well.

closed 2017-10-14 by Mayankm96

oppne, 2017-11-11:
This is the problem, i cant seem to record multiple image types, even tough i setup the json file like this?

Mayankm96, 2017-11-15:
@oppne Are you trying to record through the APIs or through the record button on the Simulator?

ganeshrk95, 2018-07-08:
I tried using API and the above code. It still records in the default resolution.
{
  "SettingsVersion": 1.2,
  "DefaultVehicleConfig":"PX4",
  "CameraDefaults": {
    "CaptureSettings": [
    {
      "ImageType": 1,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },

    {
      "ImageType": 3,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    },

    {
      "ImageType": 5,
      "Width": 640,
      "Height": 480,
      "FOV_Degrees": 90,
      "AutoExposureSpeed": 100, 
      "AutoExposureBias": 0,
      "AutoExposureMaxBrightness": 0.64,
      "AutoExposureMinBrightness": 0.03,
      "MotionBlurAmount": 0,
      "TargetGamma": 1.0
    }

  ],
  },
  "SimMode": "Multirotor",
  "AirControlCompID":1,
  "AirControlSysID":134,
  "ExtRendererCompID":1,
  "ExtRendererSysID":167,
  "ExternalSimPort":14588,
  "SimCompID":42,
  "SimSysID":142,
  "RpcEnabled": true,
  
  "Vehicles": {
      "PX4": {
      "VehicleType": "PX4Multirotor",
      "LogViewerHostIp": "127.0.0.1",
      "LogViewerPort": 14388,
      "OffboardCompID": 1,
      "OffboardSysID": 134,
      "QgcHostIp": "127.0.0.1",
      "QgcPort": 14550,
      "SerialBaudRate": 115200,
      "SerialPort": "/dev/ttyACM0",
      "SimCompID": 42,
      "SimSysID": 142,
      "SitlIp": "127.0.0.1",
      "SitlPort": 14556,
      "UdpIp": "127.0.0.1",
      "UdpPort": 14560,
      "UseSerial": true,
      "VehicleCompID": 1,
      "VehicleSysID": 135,
      "Model": "Generic",
      "LocalHostIp": "127.0.0.1"
    }
  },
  "SubWindows": [
    {"WindowID": 0, "CameraName": "0", "ImageType": 3, "Visible": true},
    {"WindowID": 1, "CameraName": "0", "ImageType": 5, "Visible": true},
    {"WindowID": 2, "CameraName": "0", "ImageType": 0, "Visible": true}    
  ],
  "Recording": {
    "RecordOnMove": false,
    "RecordInterval": 0.05,
    "Cameras": [
        { "CameraName": "0", "ImageType": 3, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "0", "ImageType": 5, "PixelsAsFloat": false, "Compress": true },
		{ "CameraName": "0", "ImageType": 0, "PixelsAsFloat": false, "Compress": true }
    ]
  }
}``

ganeshrk95, 2018-07-08:
Hi!

I am using APIs with the above code. It doesn't seem to record in the specified resolution. Any workaround?