microsoft/AirSim #531: Issue related to the orientation of drone and frame of velocity vector as drone is unable to move in the direction of its heading

shristi945, 2017-10-11:
I am controlling the drone using python script and the problem is I am not able to move the drone in the direction of its heading after a rotation with a yaw angle of 36 degrees now the heading is in slight left (if the initial heading was along x axis) yaw value is -0.6(-36 degrees) now using moveByVelocityZ keeping yaw constant does not  move in the direction of heading of drone rather it deflects more to left and yaw value changes.
want to move the drone in the direction of heading .
Here is my python script:

yaw = 0                    //initialization
pi = 3.14159265483
vx = 0
vy = 0
z=-2
change=-2*pi/10                 // changing yaw value to rotate left at angle -36 deg or yaw = -0.6
yaw = (yaw + change)
vx = math.cos(yaw)
vy = math.sin(yaw)

client.moveByVelocityZ(vx, vy,z, 1, DrivetrainType.ForwardOnly, YawMode(False, 0))  
//this would rotate to left and yaw value if printed gives correct value that is -0.6

change=0                           //keeping  change 0 and yaw angle constant
yaw = yaw + change         //yaw=-0.6
vx = math.cos(yaw)
vy = math.sin(yaw)
client.moveByVelocityZ(vx, vy,z, 1, DrivetrainType.ForwardOnly, YawMode(False, 0))  
//this deflect a bit and changes its heading inspite of moving ahead along the same angle with a yaw value of -0.6 but now the yaw value changes to -1.3.

why is this the case why cannot move in the direction of heading when yaw angle is unchanged if anyone can help than please reply.
Thanks

Kjell-K, 2017-10-12:
Remove the `YawMode(False, 0)` argument.

For me this function works. Always goes straight to heading.
```
def straight(speed):
    pitch, roll, yaw  = client.getRollPitchYaw()
    vx = math.cos(yaw) * speed
    vy = math.sin(yaw) * speed
    client.moveByVelocityZ(vx, vy, z, 1, DrivetrainType.ForwardOnly)
```

closed 2017-10-19 by sytelus

Kjell-K, 2017-10-19:
If you mean the [2bc9f72 ](https://github.com/Microsoft/AirSim/commit/2bc9f72d9f09f9f33a82cf3744445af07857a32f)commit, I tried reset from there on and I got: 

> RPCError: b"rpclib: server could not find function 'reset' with argument count 0."

But that seems as if Pose and ignorCollision are still expected. I just updated my project with update_from_git.bat and used the latest AirSimClient.py. Is that enough to get these changes or do I need to rebuild everything?

shristi945, 2017-10-24:
@Kjell-K  Thanks for the solution that works.