microsoft/AirSim #532: Reset with IgnoreCollision does not work

Kjell-K, 2017-10-11:
@sytelus I tested the new simSetPose() in my test environment. After I run some episodes, it shows that the ignore_collison feature does not work properly.

I am still colliding with obstacles on direct reset path.

```
home = client.simGetPose()
.
.
client.simSetPose(home, True);
```

Kjell-K, 2017-10-12:
To illustrate the issue I attempt to set my copter in the beginning of the script to the other side of the map. At x = 215 (Which is a free area). Attached a GIF to relate. 

Is it possible that the objects that I have created have a different setting that is not effected by your changes in IgnoreCollision?

```
    home_pos = client.getPosition()
    home_ori = client.getOrientation()
    
    client.moveToZ(-3, 3)
    
    client.simSetPose(Pose(Vector3r(215, home_pos.y_val, home_pos.z_val),
    Quaternionr(home_ori.w_val, home_ori.x_val, home_ori.y_val, home_ori.z_val)), True)
```
    
[image]

cangokalp, 2017-10-13:
I experience a difference behavior; when I use simSetPose(), it respawns at the initial position and then goes to the where it was before simSetPose(). I just want to make it respawn from the initial position and sit there till the next command - just like reset(). Does simSetPose() just changes the coordinate and make the current point set to whatever Pose given to this function, I thought it was going to take me to the Pose that I input in this case I give home inputs and expect to be transport to home, but instead I think it assigns the current point to be home. I'm very confused

cangokalp, 2017-10-13:
I experience a difference behavior; when I use simSetPose(), it respawns at the initial position and then goes to the where it was before simSetPose(). I just want to make it respawn from the initial position and sit there till the next command - just like reset(). Does simSetPose() just changes the coordinate and make the current point set to whatever Pose given to this function, I thought it was going to take me to the Pose that I input in this case I give home inputs and expect to be transported to home, but instead I think it assigns the current point to be home. I'm very confused

Kjell-K, 2017-10-13:
@cangokalp Pay attention at what point of your script you declare `home`. I just called it home and I declare it right after I arm, so in that moment home is set as starting point . .

```
    home_pos = client.getPosition()
    home_ori = client.getOrientation()

    .  Do  
    .  other 
    .  stuff. Do your episode.
 
    client.simSetPose(Pose(Vector3r(215, home_pos.y_val, home_pos.z_val),
    Quaternionr(home_ori.w_val, home_ori.x_val, home_ori.y_val, home_ori.z_val)), True)
```

sytelus, 2017-10-13:
I think there is bit of misunderstanding. the simSetPose is designed to work in "Computer Vision" mode where there is no physics. In that mode, you just set the pose for camera wherever you want. The simSetPose currently would give unpredictable behavior in multirotor mode where you do have physics. The physics engine is basically overriding whatever pose you are trying to set. I suppose it might be possible to do some hack to make this work... 

What is the purpose you are using simSetPose in multirotor mode? If it's for reset then it might be better to add reset API itself.

Kjell-K, 2017-10-14:
@sytelus Yes I know but in the at some time I got `simSetPose()` working for simple_flight mode! I am still waiting for a real reset API. 

Even `simSetPose()` does not work for me anymore when I integrate with openAI gym. So for proceeding in RL with AirSim, a real reset API as it is available for car, would be necessary. Can you help with that @sytelus?

sytelus, 2017-10-19:
I did some refactoring and reset should be working now in latest code. Binaries would be updated soon as well.

closed 2017-10-19 by sytelus

Kjell-K, 2017-10-19:
@sytelus  If you mean the 2bc9f72 commit, I tried reset from there on and I got:

> RPCError: b"rpclib: server could not find function 'reset' with argument count 0."

But that seems as if Pose and ignorCollision are still expected. I just updated my project with update_from_git.bat and used the latest AirSimClient.py. Is that enough to get these changes or do I need to rebuild everything?